SCADA-LTS/Scada-LTS #52: Angular JS as optional Dashboard

tominagaf, 2016-10-18:
Mango M2M has an insteresting feature called Custom Dashboards which is based on Angular JS interface. You can see it here https://vimeo.com/179788325 

It would be really nice having this dashboard.

grzesiekb, 2016-10-18:
Welcome!
We have a task management hosted on JIRA cloud (https://scadalts.atlassian.net). Please provide us email to send you invitation. Are you going to participate or contribute into this feature?

tominagaf, 2016-10-18:
Hello, 
We have been talking by email. I am just an advanced user, but I can contribute on testing and whatever you need. I am teaching myself coding but I am a begginer. 

Please send the invitation to this email: webmonitor@outlook.com

closed 2016-10-19 by tominagaf