SCADA-LTS/Scada-LTS #92: java.lang.Exception: Unable to derive image dimensions

SimSonic, 2016-12-21:
Hello.

I've tried to run ScadaBR and Scada-LTS (latest "release" from here) under Tomcat 9 (and 8 too) but everytime I receive this messages in tomcat9-stdout.2016-xx-xx.log:

```
2016-12-21 15:31:33 Commons Daemon procrun stdout initialized
INFO  2016-12-21 15:31:47,558 (com.serotonin.mango.MangoContextListener.contextInitialized:88) - Mango context starting 
WARN  2016-12-21 15:31:47,716 (com.serotonin.mango.view.ViewGraphicLoader.loadViewGraphics:58) - Failed to load image set at D:\Temporary\ApacheTomcat9\webapps\Scada-LTS\graphics\Anti-horario-on-off 
java.lang.Exception: Unable to derive image dimensions
	at com.serotonin.mango.view.ViewGraphicLoader.loadDirectory(ViewGraphicLoader.java:119)
	at com.serotonin.mango.view.ViewGraphicLoader.loadViewGraphics(ViewGraphicLoader.java:55)
	at com.serotonin.mango.MangoContextListener.imageSetInitialize(MangoContextListener.java:505)
	at com.serotonin.mango.MangoContextListener.contextInitialized(MangoContextListener.java:105)
	at org.apache.catalina.core.StandardContext.listenerStart(StandardContext.java:4717)
	at org.apache.catalina.core.StandardContext.startInternal(StandardContext.java:5179)
	at org.apache.catalina.util.LifecycleBase.start(LifecycleBase.java:183)
	at org.apache.catalina.core.ContainerBase.addChildInternal(ContainerBase.java:752)
	at org.apache.catalina.core.ContainerBase.addChild(ContainerBase.java:728)
	at org.apache.catalina.core.StandardHost.addChild(StandardHost.java:734)
	at org.apache.catalina.startup.HostConfig.deployWAR(HostConfig.java:952)
	at org.apache.catalina.startup.HostConfig$DeployWar.run(HostConfig.java:1823)
	at java.util.concurrent.Executors$RunnableAdapter.call(Unknown Source)
	at java.util.concurrent.FutureTask.run(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.lang.Thread.run(Unknown Source)
```
for every folder name in ScadaBR\graphics\<dirs>.

OS: Windows 7 x64.
Java: Oracle 8u111.
Tomcat: 8 and 9.

Thanks.

seangordon, 2016-12-21:
I saw this same error previously, it is related to running on Tomcat 8.

If you run on Tomcat 7 you do not see this error (I don’t think that ScadaLTS has been ported to Tomcat 8, and I know that ScadaBR only runs on Tomcat 7).

grzesiekb, 2016-12-21:
For latest release.
We use 
  OS: (Linux / Windows),
  Tomcat: 7, 
  Java: (Oracle - jdk 1.8  / open-jdk 1.8),
  SQL: mysql 5.6.

grzesiekb, 2017-04-20:
Run ScadaLTS on docker - [wiki](https://github.com/sdtabilit/Scada-LTS/wiki/Run-ScadaLTS-on-docker---instruction)

SoftQ, 2018-09-06:
Hello, 
this issue has been fixed  in the following commit:
 https://github.com/SCADA-LTS/Scada-LTS/commit/ba32e610a99253c32342b69e6d4818ca40af3b83

radek2s, 2021-01-15:
This bug has been fixed in followed commit [c6efb02f88fbaef9d3e5b4d839a67a8a246f1d0f](https://github.com/SCADA-LTS/Scada-LTS/commit/c6efb02f88fbaef9d3e5b4d839a67a8a246f1d0f) on a related [branch](https://github.com/SCADA-LTS/Scada-LTS/tree/feature/%2392_fix_image_dimensions_error_log). Now Images for Scada-LTS works both in Tomcat7 and Tomcat8 and that log error do not exists any more. 

It is a problem with line **99** in _ViewGraphicLoader.java_ where `String imagePath = file.getPath().substring(path.length());` **path.length()** has differ between this two version of Tomcat.

closed 2021-01-22 by radek2s