SCADA-LTS/Scada-LTS #273: NullPointerException

semirov, 2017-05-05:
When you add any register in the serial modbus source, an exception is thrown com.serotonin.modbus4j.exception.ModbusTransportException: java.lang.NullPointerException
[image]

war version - v0.0.7-next-test

grzesiekb, 2017-05-05:
Hello, thank You for submitting information. With regard to maximizing the probability of solving Your problem, please:
- write it to StackOverflow with "scadalts" tag
- please test/try to replicate the problem on newest Docker image - Scada-LTS has newer Modbus version [https://github.com/SCADA-LTS/Scada-LTS/wiki/Run-ScadaLTS-on-docker---instruction](https://github.com/SCADA-LTS/Scada-LTS/wiki/Run-ScadaLTS-on-docker---instruction)

CHNEG8805, 2018-06-10:
hello @grzesiekb i use the last version of scadalts and i meet the same problems. how can i solve it ?

Arruelah, 2018-12-03:
> hello @grzesiekb i use the last version of scadalts and i meet the same problems. how can i solve it ?

Same problema here. No nodes appear, but other software, using same converter, and port work normally. Seems like tomcat has no access over the port. The RX LED on the PLC does not blink when ScadaLTS tries to communicate.

CHNEG8805, 2018-12-03:
Seems to be the a problem of driver of TXRX 。 Suggested Use the driver for ARM

fabiodurao, 2019-01-30:
The ScadaLTS came from the ScadaBR 1.1 branche. This problem has been inherited, below is the link of the ScadaBR forum where a user has solved the Modbus Serial problem. I am an Instructor of the Training with ScadaBR / ScadaLTS in Brazil.

http://forum.scadabr.com.br/t/erro-no-driver-modbus-serial-na-versao-do-scadabr-1-1-0-rc-do-branches/1307/9

lucas-silva-assy, 2019-11-25:
Good afternoon everyone,
Any solutions to the problem? How to build the problem class as Fabio said?

Limraj, 2023-06-03:
Hi @CHNEG8805, @lucas-silva-assy, @germanorocha, @Arruelah, @semirov,
The problem with Modbus Serial should not occur from version 2.7.2:
https://github.com/SCADA-LTS/Scada-LTS/issues/1694
However, this may not work on ARM processors.

The recommended version is the latest version, currently 2.7.5.3. Stable version 2.6.18 does not have this fix. In the near future we plan to release a stable version from version "2.7.5.*".

Regards,
Kamil Jarmusik

closed 2023-06-03 by Limraj