SCADA-LTS/Scada-LTS #353: Test version 0.0.8

ArkadiuszParafiniuk, 2017-09-07:


ArkadiuszParafiniuk, 2017-10-12:
### **Test - disable/enable data source with points with event handler in context**
[Click here for test scenario](https://github.com/SCADA-LTS/Scada-LTS/issues/374)
**Result: 5/5 - passed ✔**

| Test Step        | Test Data           | Expected Result  | Result  |
| ------------- |:-------------:| :-----:|:-----:|
| 1. | Log as administrator. Go to "Data Source" tab. Create data "Virtual Data Source" with name "test_ds" and enable ds. Create data point with name="vd_test_binary", Settable="true", Data type="Binary", set settable true, Change type="No change", Start value="0" . Then save data point and Enable data point  | The data point should be created properly. |✔|
| 2. |  Go to "Data Source" tab and details data point  "vd_test_binary" tab. Create Even detectors  Then save data point  | The data point should be have event detector properly. |✔|
| 3. | Go to "Data Source" tab and disable data source "test_ds". | You should see data point "vd_test_binary" disabled with EventDetecor|✔|
| 4. | Go to "Data Source" tab and turn on data source. | Data source should be enabled again |✔|
| 5. | Go to "vd_test_binary" data point properties. | Data point should still have event detector |✔|

ArkadiuszParafiniuk, 2017-10-12:
### **Test - Event Handler saving**
[Click here for test scenario](https://github.com/SCADA-LTS/Scada-LTS/issues/229)
**Result: 2/2 - passed ✔**

| Test Step        | Test Data           | Expected Result  | Result  |
| ------------- |:-------------:| :-----:|:-----:|
| 1. Open data source tab and press show points icon and show properties | Add event detector with high limit type. Set High limit to 5 and duration to 2 minutes. Save | After one second schould be avalible new Event detector with Test Data values |✔|
| 2. Data point still has an event detector after rewriting the point. |  Save datapoint again. Go to different view and back to previous. | Event detector should still exist at data point properties |✔|

grzesiekb, 2017-10-12:
### **Test - User access to data points and graphical views V2**
[Click here for test scenario](https://github.com/SCADA-LTS/Scada-LTS/issues/351)
**Result: 2/5 - failed ❌**

| Test Step        | Test Data           | Expected Result  | Result  |
| ------------- |:-------------:| :-----:|:-----:|
| 1. | Log as administrator. Go to "Users profiles" tab. Create user profile with name "test_profile". Make sure that data source "Test_meta" is unchecked, give permission "set" to graphical view "test_view", and give permission "set" to watch list "test" . Then save user profile.  | The users profile should be created properly. | ✔|
| 2. |  Go to "Users" tab. Create new no administrator user with user profile "test_profile".  | The user should be created properly. | ✔|
| 3. | Log in as created no administrator user. Go to graphical views tab. | You should see graphical view "test_view" with 1 simple point component(value of point=4)  | ❌ Can not access to "test_view" |
| 4. | Try to change value of point to 5  | Value of point should change properly to 5 | ❌ As above|
| 5. | Go to watch list tab and select watch list "test" from the list | Watch list should display properly and it shoud be possible to set point | ❌ Can not access to watch list "test" |

grzesiekb, 2017-10-12:
### **Test - watchlist**
[Click here for test scenario](https://https://github.com/SCADA-LTS/Scada-LTS/issues/284)
**Result: 1/25 - passed 1**

| Test Step        | Test Data           | Expected Result  | Result  |
| ------------- |:-------------:| :-----:| :-----:|
| 1. | Log as administrator. Click tab „Watchlist”. | Open Watchlist.|✔|
| 2. | Project: Data source. Press „add” and then go back and click again to „Data source” | Move to data source list |✔|
| 3. | Project: Data Sources. Change field „1-wire” to „Virtual Data Source” Press „save” | A message appears (in red colour): ”Name is required” |✔|
| 4. | Project: Data Sources. Change field „1-wire” to „Virtual Data Source”. Click „add”. Name „Test VDP” in field „update period” type 1 second. Click save and enabled. Press button „edit properities”.Name „Test-Numeric”. Data type change to „numeric”. Change type to „random” minimum 0, maximum 100, start value 1. Click save. Change status to enabled. Go to  „Old watchlist” on the list press „TestVDP – Test-numeric” | After „press button „load” in the tab „watchlist” the Virtual data Source „Test-numeric” appears on the chart, randomly changes every second itself value between 0 to 100. |❌ Can not access rxtx DLL or libraties have been correctly installed |
| 5. | Go to „Data Source” change status „TestVDP” from „enabled” to „disabled” | Go to „Data Source” change status „TestVDP” from „enabled” to „disabled” |❌ As above|
| 6. | Go to „Old watchlist” click on the icon „delete from the watchlist” | „Test-Numeric” chart stopped. |
| 7. | Go to „Old watchlist” on the „TestVDP – Test-numeric”. | „Test-numeric” chart correctly run, after load chart on the „watchlist”. |
| 8. | Click button „Sign out” and again log as admin. Go to watchlist and click button „Reveal chart” | „Test-numeric” chart run correctly |
| 9. | „Test-numeric” chart run correctly. | On the bottom appears message „URL Copied to the clipboard”. |
| 10. | Click icon „increase line width” | Line width correctly increase |
| 11. | Click icon „decrease line width” | Line width correctly decrease |
| 12. | Click icon „toggle line shape” | New line shape is visible and change correctly. |
| 13. | Click 3 times icon „toggle line shape” | New line shapes is visible and changes correctly |
| 14. | Click icon „Toggle line mode”. | New line mode is visible and change correctly.  |
| 15. | Click icon „Toggle legend”. | Legend disapear. |
| 16. | Again click icon „Toggle legend” | Legend appears on the bottom |
| 17. | Click icon „Toggle fullscreen” | Chart has been enlarged |
| 18. | Again click icon „Toggle fullscreen”. | Chart toggle to standard mode. |
| 19. | Click icon „Toggle logarithmic Y axis”. | Chart correctly change to logarithimc Y axis. |
| 20. | Click icon „download plot as png”. | Plot has been correctly save as png. |
| 21. | Go to „Old watchlist” click on the icon „edit watch list name”. Name: „Test1”, click save. | New watch list has been added |
| 22. | Add „TestVDP – test-numeric” to the new „test1” watchlist. | Data point „Test-numeric” now belongs to watch list „test1” |
| 23. | Click icon „copy watch list”. | New watch list is visible as „Copy of the test1 ”. |
| 24. | Click icon „Add new watch list”. | New watch list has been created. It is name as „unnamed”. |
| 25. | Click icon „Delate”. | „unnamed” watchlist has been succesfully deleted. |

ArkadiuszParafiniuk, 2017-10-13:
### **Test - Display of warnings depending on alarm level**
[Click here for test scenario](https://github.com/SCADA-LTS/Scada-LTS/issues/390)
**Result: 5/5 - passed ✔**

| Test Step        | Test Data           | Expected Result  | Result  |
| ------------- |:-------------:| :-----:|:-----:|
| 1. Create virtual data source | Log as administrator. Go to data sources. Create new virtual data source "testDS" with 1 second update period. Turn on data source. | Data source should be created and enabled. |✔|
| 2. Create data point | At "testDS" create virtual data point "testDP"(numeric data type, random change type, minimum 0, maximum 3, start 0). Save and turn on data point. |  Data point should be created and enabled. |✔|
| 3. Set event detectors | For "testDP" add 2 "high limit" event detectors.  First should have high limit 1 and alarm level "none". Secound should have high limit 2 and alarm level "information". Save | Data point "test" should have 2 "high limit" event detectors |✔|
| 4. | Go to watchlist. Add data point  "testDP" to watchlist. | On alarm list should be display only warnings with alarm level "information"(yellow flag) |✔|
| 5. | Go to graphical views. Create new graphical view "testGV" and add simple point component for "testDP". | You should see yellow "warnings" triangle on the left side of simple point component. When cursor is on triangle you should see alarm list with only warnings with "information" level alarms. |✔|

ArkadiuszParafiniuk, 2017-10-13:
### **Test - Language Change**
[Click here for test scenario](https://github.com/SCADA-LTS/Scada-LTS/issues/236)
**Result: 1/11 - failed ❌**

| Test Step        | Test Data           | Expected Result  | Result  |
| ------------- |:-------------:| :-----:|:-----:|
| 1. English | Set language to english from menu upper left corner glob icon. Pass one after another all sub page from menu | Massage and information on sub page displayed correctly in indicated language. On page not displayed markers "?text?". In all massage displayed diactric marks. |✔|
| 2. German | Repeat step 1 for German language(Deutsch) | As above |❌    Some words have not been translated from English|
| 3. Russian | Repeat step 1 for Russian language(Руссиан) | As above |❌    As above|
| 4. Finnish | Repeat step 1 for Finnish language(Suomi) | As above |❌    As above|
| 5. Portuguese | Repeat step 1 for Portuguese language(Português) | As above |❌    As above|
| 6. Luxembourg | Repeat step 1 for Luxembourg language(Luxembourgeois) | As above |❌    As above|
| 7. French | Repeat step 1 for French language(Francais) | As above |❌    As above|
| 8. Chinese | Repeat step 1 for Chinese language(中文) | As above |❌    As above|
| 9. Dutch | Repeat step 1 for Dutch language(Nederlands) | As above |❌    As above|
| 10. Spanish | Repeat step 1 for Spanish language(Español) | As above |❌    As above|
| 11. Polish | Repeat step 1 for Polish language(Polski) | As above |❌    As above|

ArkadiuszParafiniuk, 2017-10-13:
### **Test - access to the data point details for non-admin users**
[Click here for test scenario](https://github.com/SCADA-LTS/Scada-LTS/issues/393)
**Result: 6/6 - passed ✔**

| Test Step        | Test Data           | Expected Result  | Result  |
| ------------- |:-------------:| :-----:|:-----:|
| 1. Create virtual data source | Log as administrator. Go to data sources. Create new virtual data source "testDS" with 1 second update period. Turn on data source. | Data source should be created and enabled. |✔|
| 2. Create data point | At "testDS" create virtual data point "testDP"(numeric data type, random change type, minimum 0, maximum 3, start 0). Save and turn on data point. |  Data point should be created and enabled. |✔|
| 3. | Go to graphical views. Create new graphical view "testGV" and add simple point component for "testDP". | You should see yellow "warnings" triangle on the left side of simple point component. When cursor is on triangle you should see alarm list with only warnings with "information" level alarms. |✔|
| 4. | Go to "user profiles, create new user profile "test_profile" and give it access rights to graphic view "testGV". Save | User profile should be created |✔|
| 5. | Go to users. Create new user "user" (enabled, without administrator rights, with user profile "test_profile"). Save| User account should be created |✔|
| 6. | Log in as "user", go to graphical view "testGV", hover your cursor on "testDP" and click on cube in left top corner | You should be redirected to data point details page |✔|

ArkadiuszParafiniuk, 2017-10-13:
| Test Step        | Test Data           | Expected Result  |
| ------------- |:-------------:| :-----:|
| 1. Go to sub page "User" | Click icon "Users" | Open page 'Users" |
| 2. | Click icon "Add new User" | Open menu create new user "User details" |
| 3. Empty field | Leave all field empty. Click button "Save" | Can`t create new user. Massage appears "Required value" in fields "Username", "New password" and "Email" |
| 4.Empty password and Email | Enter only "Username" for example "test1" Click button "Save" | Massage appears "Required value" in fields "New password" and "Email" |
| 5. Empty email | Enter 'User Name' and 'Password'. For example User name: Test1, Password: test | Massage appears "Required value" in "Email". Click button "Save" |
| 6. Correct data | Enter "User Name" Test1, Password test, Email; test@test.com Click button "Save" | Correctly add new user. Massage appears "User added" |
| 7. The same username | Click button "Add user". Enter user name Test1. Click button "Save" | Can`t create new user, massage appears "Username must be unique" |
| 8. User with Administrator permission | Click button "Add user". Enter user name testA Password test email testa@test.com. Mark checkbox "Administrator" Click button "Save" | Create user with administrator permission |
| 9. Disable user | Click button "Add user". Enter user name testD Password test email testd@test.com. Mark checkbox "Disabled" Click button "Save". | Create new user with disable account |
| 10. | Click "Sign out" button | Move to login page |
| 11. Log in to new create account | Enter "User id" Test1 ,"Password" test Click button "Login" | Correctly log in to the system. Available only 8 sub page ("Watch list", "Watchlist edit", "Old watchlist", "Views", "Alarms", "Reports", "Users", "System") |
| 12. | Click button "Sign out" | Move to login page |
| 13. Log in as Administrator | Enter "User id" TestA ,"Password" test Click button "Login" | Correctly log in to the system with full administrator permission |
| 14. | Click "Sign out" button | Move to login page |
| 15. Log in as disable user | Enter "User id" TestD ,"Password" test Click button "Login" | 	Can`t log in to the system massage appears "Your account has been disabled. Please contact your administrator" |

ArkadiuszParafiniuk, 2017-10-13:
| 1. Create new test folders in empty hierarchy | Open Point Hierarchy tab. Add 5 different folders (for ex. TestF1 ... TestF2) | Five folders are visible at basic level of hierarchy. |
| 2. | Refresh view (Shift - Ctrl - R) | All empty folder should still be visible on a basic lavel. In database (pointhierarchy table) should looks like: <br> / id / parentId / name <br> / 1 / 0 / TestF1 <br> / 2 / 0 / TestF2 <br> / 3 / 0 / TestF3 <br> / 4 / 0 / TestF4 <br> / 5 / 0 / TestF5 |
| 3. Create test datapoints which are nodes in Point Hierarchy | Create new Virtual DataSource. Create and delete 3 data points and next add 5 datapoints (TestDP1 ... TestDP5) Open Point Hierarchy tab. | Under list of 5 folders in basic level of hierarchy schould apper our 5 datapoints. |

grzesiekb, 2017-10-16:
| Test Step        | Test Data           | Expected Result  |
| ------------- |:-------------:| :-----:|
| 1. | Log as admin. go to "Dashboard". Click tab "Views". Click tab "edit view". | Should appears informations "Graphic view not defined". |
| 2. | Log as admin. Click tab "Graphic views".Click icon "Add view" | Is displayed graphic view page |
| 3. | Click on the button "New view". | Is displayed view properties |
| 4. | Click button "save" | Next to"name" field appears (in red colour) information "Required value"|
| 5. | Change "Export ID (XID)" field to empty space. Click save.  | Next to"Export ID (XID)" field appears (in red colour) information "Required value". |
| 6. | Click icon "Choose file". Choose random background image and click "Upload image". | Background image correctly appears in view window. |
| 7. | Go to "Graphic view". Click icon "Edit view". Click button "Clear image". | Background image has been correctly delated. |
| 8. | Open "Graphic view". Click icon "Add view". Change resolution to "640x480". "Components: Alarms List". Click button "Add component to view". Click button "save". | On the "Graphic view", should appears "Alarms list" in window resolution 640x480. |
|  |  |  |
|  |  |  |
|  |  |  |
|  |  |  |
|  |  |  |
|  |  |  |
|  |  |  |
|  |  |  |
|  |  |  |

grzesiekb, 2017-10-16:
| Test Step        | Test Data           | Expected Result  |
| ------------- |:-------------:| :-----:|
| 1. | Go to "Settings”: "import/export”. In the first section click icon "Help”, nex to word "Export"/ | In dialog box appears "Importing and Exporting configuration". |
| 2. | Click on the second „Help” icon. It is next to "Export Project (Download)" sentence. | Overview |
| 3. | Click third „Help” icon on the bottom.  It is above sentence: „Select the file (.zip) e click the button to import it”. | As above |
| 4. | Go to "watchlist". Click icon "Help", next to "Points". | In dialog box appears "watchlist points". |
| 5. | Click icon "Help", next to "watch lists". | In dialog box appears "watchlist". |
| 6. | Click icon "Help", next to "Chart". | In dialog box appears "watch list charts". |
| 7. | Go to "Graphic views". Click icon "Help", next to "Graphic views". | In dialog box appears "Graphical views". |
| 8. | Click icon "New view". Click icon "Help", next to "View properties". | In dialog box appears "Editing graphical views". |
| 9. | Go to "Reports". Click icon "Help", next to "Report queue". | In dialog box appears "Reports instance". |
| 10. | Click icon "Help", next to "Report templates". | In dialog appears "Report templates" . |
| 11. | Go to "Event handlers" and click icon "Help" next to "Event handlers". | In dialog appears "Event handlers". |
| 12. | Go to "Data sources". Click icon "Help", next to "Data sources". | In dialog appears "Data source lists". |
|  |  |  |
|  |  |  |
|  |  |  |
|  |  |  |
|  |  |  |
|  |  |  |

grzesiekb, 2017-10-16:
Nr. 13
### **Login Controller(new UI)**
[Click here for test scenario](https://github.com/SCADA-LTS/Scada-LTS/issues/240)
**Result: 0/9 - passed**

| Test Step        | Test Data           | Expected Result  |
| ------------- |:-------------:| :-----:|
| 1. Correct UI |  |  |
| 2. Try to enter to page without login | Try go to page watch list ("/#/appBody/watchlist") by manually typing in browser | You will be move to login page |
| 3. Empty space | Leave "Username" field empty and enter any value in "Password" field | A message appears ( in red colour) "Enter username and password" |
| 4. Empty space | Leave "Password" field empty, and enter any value in "Username". Try to log in. | A message appears ( in red colour) "Enter username and password" |
| 5. Empty space | Leave both field empty. try to log in | A message appears ( in red colour) "Enter username and password" |
| 6. Not existing user | Enter any value in "Username" and "Password" field | A message appears "Wrong username or password" |
| 7. Wrong password | In "Username" enter "admin" in "Password" field enter any value. Try log in | A message appears "Wrong username or password" |
| 8. Correct password and user id | Enter in "Username" admin and "Password" admin | Log in correctly and move to "/#/appBody" page |
| 9. Log out | Click button "Sign Out" | Moved to log in page |

closed 2017-10-16 by grzesiekb

ArkadiuszParafiniuk, 2017-11-02:
| Test Step        | Test Data           | Expected Result  |
| ------------- |:-------------:| :-----:|
| 1. | Go to "Scripting" page | Redirection to "scripting page" |
| 2. | Click on "Add a new script". Save. | A message appears (in red colour): ”Name is required” |
| 3. | Enter a name "testScript". Save | Script should be added properly. |
| 4. | Add "meta_point" from context objects list. Enter script value: "point='DP_391319'; if(p1.value > 3) {  dp.writeDataPoint(point,1); } else { dp.writeDataPoint(point,0); }".  Add datapoints commands with var "dp". Save | A message appears (in red colour): ”Script saved” |
| 5. | Click button "Run" (gear icon). | A message appears (in red colour): ”Executed successfully” |
| 6. | Go to data sources page, "result Point" point details. | "resultPoint should have value 1 (true)" |