SCADA-LTS/Scada-LTS #423: Getting Wierd Error

abhishekricky, 2017-11-20:
WARN  2017-11-21 01:42:38,568 (com.serotonin.mango.rt.dataSource.PollingDataSource.scheduleTimeout:63) - Panel 3A: poll at 2017/11/21 01:42:38 aborted because a previous poll started at 2017/11/21 01:42:08 is still running 

What can be the issue ?

grzesiekb, 2017-11-21:
We need access to the installation to diagnose it.

andrebarradas, 2022-05-09:
Hi guys, i have the same problem, any suggestion?

`WARN  2022-05-09 16:37:41,296 (com.serotonin.mango.rt.dataSource.PollingDataSource.scheduleTimeout:63) - GATEWAY: poll at 2022/05/09 16:37:41 aborted because a previous poll started at 2022/05/09 16:36:03 is still running `

Limraj, 2024-02-29:
Hi @abhishekricky, @andrebarradas, @grzesiekb,

The reason for this message is related to Polling Data Source, i.e. Data Sources that have the update parameter. Now if the current update (calling the scheduleTimeout method and then doPoll) does not execute before the next update, this thread is stopped, with this message. This means that if the update execution time is longer than the period set in the Data Source, it will be updated as often as physically possible. This appears to be correct behavior. 

Regards,
Kamil Jarmusik

closed 2024-02-29 by Limraj