SCADA-LTS/Scada-LTS #459: Unable to display the text output of server side script

TomyZacharia, 2018-01-31:
We have started using scadalts recently. We have used scadabr in the past and the problem did not exist.

The following problem is observed.
1. Input point defined as Modbus IP - Data is read as "Coil Status" and binary.  It is available and displaying in watch list as well as Graphical Views
2. This point is used as input to Meta Data source. The point is defined as AlphaNumeric and script has been written to interpret the status and display as either "OFFLINE" or "ONLINE"
3. The output of metadata script is evaluated correctly and is displayed by "Validate Script" button.
4. It is displayed incorrectly in watch list as "(n/a)" and in the graphical view as "null".

We are using ubuntu 16.04, Tomcat 7, mysql 5.7 and openjdk 1.8.

grzesiekb, 2018-02-01:
I need detailed information that allows me to replicate the indicated situation - For example - script described in the second point.

TomyZacharia, 2018-02-01:
I am attaching screen shots for each point.
1. screen shot for input data
2. screen shot for the script
3. screen shot of watchlist
4. screen shot of graphical view
[image]
[image]
[image]
[image]

grzesiekb, 2018-02-01:
Thank you, I will check it today :) - 
If the modbus value changes, then the value in meta data source will be calculated (in this case). 
It may not necessarily be the reason, I will check it later.

grzesiekb, 2018-02-02:
### (EN) Unable to display (alphanumeric) the text output of server side script.
#### Thank you for your application and valuable comments.

The above method for calculating the value and its checking is not very optimal.

I suggest you configure the program a little differently.

Namely, the point type set as in Meta data source as:
**Multistate**

[image]

Then change the way text is rendered

[image]

What will work properly.

[image]

[image]

Thank you.

TomyZacharia, 2018-02-10:
We were able to implement the same successfully. We have used graphic element type "simple point" for implementing the same. Thanks for the reply.

We further need the process for changing the formatting of graphic element "simple point". A button is available with the label "style attribute". Screen shot is attached. We require the settings for changing font size, bold/italic font type, change text box boundary color etc.

[image]

radek2s, 2018-02-12:
Alphanumeric data type has been fixed. Now it can display a message with a custom return value from users custom script. It works also in WatchList and View menu. 

Due to styling the "simple point" element here are some examples how it works.
[image]

closed 2018-02-15 by radek2s