SCADA-LTS/Scada-LTS #465: Unable to export project from SCADALTS

TomyZacharia, 2018-02-13:
We have a fairly large project, which we are in the process of migration from ScadaBR 0.9. We finally managed to migrate by following the procedure as follows.
1. Exporting only Watch lists, Graphical views, Data sources and Data points. 
2. Create users in SCADALTS with same names manually. (Also works by making all users administrators prior to export)
3. We had to manually copy and change permissions for Graphics and Uploads folders.
4. Import the JSON file created in step 1. (Alternatively change users from "administrators" to normal)
[image]

The problem we are facing is of export project from the SCADALTS project. We are getting the attached error screen whenever we carry out an export.
[image]
 
The export works if all the users are made "Admin" users.

We are using ubuntu 16.04, Tomcat 7, mysql 5.7 and openjdk 1.8. SCADA LTS v.0.0.9

grzesiekb, 2018-02-13:
###  It is quite a big topic that requires designing and simplification in my opinions.

  * The import and export mechanism  based on an external library to which we don't have source which causes many problems.
  * Another surprise is that it is not possible to operate on the database, because the data is stored in blobs that are serialized.
  * No security of data inconsistencies resulting from the use of id and Xid.

E.g
Separating individual modules and then preparing data export with their own REST API and databases (designed in accordance with art).

SoftQ, 2018-05-22:
Hello Tomy
Thank You for feedback. At present we are working on fixing datapoint hierarchy export/import. In case You have any needs or additional questions please don't hesitate to write us email to info@scada-lts.org

closed 2018-05-22 by SoftQ