SCADA-LTS/Scada-LTS #472: HTTP Receiver Data Source

abhishekricky, 2018-02-23:
After 0.0.9.1 Update, Alphanumeric values are being stored properly in HTTP Receiver Data Source 

But using timestamp tag like & __time is not working

grzesiekb, 2018-02-24:
What you mean by (like & __time is not working)? Thank you in advance for you help.

justindressler, 2019-06-21:
Hi I was able to replicate this issue the __time parameter does not override the time the change is recorded.
So if I POST or GET the httpds end point it records the data point value at the time received not the time provided by the __time parameter.

justindressler, 2019-06-21:
This also effects other data sources at least DNP. Events should be recorded with the timestamp from the source is specified.

waqassiddiqi, 2019-06-24:
I tested it on latest version (`/new/develop`) and it seems to be working.

closed 2023-05-02 by Limraj