SCADA-LTS/Scada-LTS #476: Cannot add any data source, "error" popup

evranch, 2018-02-26:
Running on Linux Mint Sylvia (Ubuntu Xenial underlying system) and installed as specified here https://github.com/SCADA-LTS/Scada-LTS/wiki/Install-Scada-LTS-on-linux

I've tried both an install in /opt and also the latest docker container. The same thing happens on attempting to add any type of data source, I get a popup simply stating "error" when I click the save icon.

grzesiekb, 2018-02-27:
Try using the docker - https://github.com/SCADA-LTS/Scada-LTS/wiki/Run-ScadaLTS-on-docker---instruction

evranch, 2018-02-27:
Exact same error with the docker, a blank window with just "error" in it. Other parts of the interface seem to work fine, i.e. I can create scheduled events, graphical views etc. 

Running from my local install because I know where to find the logfiles,

Here are the relevant logs as far as I can see

dao.log
```
DEBUG 2018-02-27 10:20:56,067 (com.serotonin.mango.web.dwr.DataSourceEditDwr.tryDataSourceSave:245) - Trying to save datasource TEST 
DEBUG 2018-02-27 10:20:56,067 (com.serotonin.mango.rt.RuntimeManager.saveDataSource:285) - Stoping DS: TEST 
DEBUG 2018-02-27 10:20:56,067 (com.serotonin.mango.rt.RuntimeManager.saveDataSource:288) - DS Stoped! 
DEBUG 2018-02-27 10:20:56,067 (com.serotonin.mango.rt.RuntimeManager.saveDataSource:292) - Saving DS: TEST
```
mango.log

`ERROR 2018-02-27 10:20:56,068 (com.serotonin.mango.web.filter.ExceptionDetectionFilter.doFilter:52) - DWR invocation exception 
java.lang.AbstractMethodError: Method com/mysql/jdbc/ServerPreparedStatement.setBlob(ILjava/io/InputStream;)V is abstract
	at com.mysql.jdbc.ServerPreparedStatement.setBlob(ServerPreparedStatement.java)
	at sun.reflect.GeneratedMethodAccessor78.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.apache.tomcat.jdbc.pool.interceptor.AbstractQueryReport$StatementProxy.invoke(AbstractQueryReport.java:212)
	at com.sun.proxy.$Proxy11.setBlob(Unknown Source)
	at sun.reflect.GeneratedMethodAccessor78.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.apache.tomcat.jdbc.pool.interceptor.AbstractQueryReport$StatementProxy.invoke(AbstractQueryReport.java:212)
	at com.sun.proxy.$Proxy11.setBlob(Unknown Source)
	at sun.reflect.GeneratedMethodAccessor78.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.apache.tomcat.jdbc.pool.StatementFacade$StatementProxy.invoke(StatementFacade.java:114)
	at com.sun.proxy.$Proxy11.setBlob(Unknown Source)
	at sun.reflect.GeneratedMethodAccessor78.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at net.bull.javamelody.JdbcWrapper$StatementInvocationHandler.invoke(JdbcWrapper.java:145)
	at net.bull.javamelody.JdbcWrapper$DelegatingInvocationHandler.invoke(JdbcWrapper.java:286)
	at com.sun.proxy.$Proxy12.setBlob(Unknown Source)
	at com.serotonin.db.spring.ArgTypePreparedStatementSetter.setValues(ArgTypePreparedStatementSetter.java:57)
	at com.serotonin.mango.db.dao.BaseDao$1.createPreparedStatement(BaseDao.java:122)
	at org.springframework.jdbc.core.JdbcTemplate.execute(JdbcTemplate.java:583)
	at org.springframework.jdbc.core.JdbcTemplate.update(JdbcTemplate.java:849)
	at com.serotonin.mango.db.dao.BaseDao.executeInsert(BaseDao.java:133)
	at com.serotonin.mango.db.dao.BaseDao.doInsert(BaseDao.java:150)
	at com.serotonin.mango.db.dao.DataSourceDao.insertDataSource(DataSourceDao.java:161)
	at com.serotonin.mango.db.dao.DataSourceDao.saveDataSource(DataSourceDao.java:119)
	at com.serotonin.mango.rt.RuntimeManager.saveDataSource(RuntimeManager.java:293)
	at com.serotonin.mango.web.dwr.DataSourceEditDwr.tryDataSourceSave(DataSourceEditDwr.java:246)
	at com.serotonin.mango.web.dwr.DataSourceEditDwr.saveVirtualDataSource(DataSourceEditDwr.java:414)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.directwebremoting.impl.ExecuteAjaxFilter.doFilter(ExecuteAjaxFilter.java:34)
	at org.directwebremoting.impl.DefaultRemoter$1.doFilter(DefaultRemoter.java:428)
	at com.serotonin.mango.web.dwr.util.LoggedInAjaxFilter.doFilter(LoggedInAjaxFilter.java:39)
	at org.directwebremoting.impl.DefaultRemoter$1.doFilter(DefaultRemoter.java:428)
	at com.serotonin.web.dwr.LocalizationFilter.doFilter(LocalizationFilter.java:42)
	at org.directwebremoting.impl.DefaultRemoter$1.doFilter(DefaultRemoter.java:428)
	at com.serotonin.mango.web.filter.ExceptionDetectionFilter.doFilter(ExceptionDetectionFilter.java:40)
	at org.directwebremoting.impl.DefaultRemoter$1.doFilter(DefaultRemoter.java:428)
	at org.directwebremoting.impl.DefaultRemoter.execute(DefaultRemoter.java:431)
	at org.directwebremoting.impl.DefaultRemoter.execute(DefaultRemoter.java:283)
	at org.directwebremoting.servlet.PlainCallHandler.handle(PlainCallHandler.java:52)
	at org.directwebremoting.servlet.UrlProcessor.handle(UrlProcessor.java:101)
	at org.directwebremoting.servlet.DwrServlet.doPost(DwrServlet.java:146)
	at javax.servlet.http.HttpServlet.service(HttpServlet.java:650)
	at javax.servlet.http.HttpServlet.service(HttpServlet.java:731)
	at org.apache.catalina.core.ApplicationFilterChain.internalDoFilter(ApplicationFilterChain.java:303)
[... 94 lines cut ...]
	at org.apache.catalina.core.ApplicationFilterChain.internalDoFilter(ApplicationFilterChain.java:241)
	at org.apache.catalina.core.ApplicationFilterChain.doFilter(ApplicationFilterChain.java:208)
[... cut ...]
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at org.apache.tomcat.util.threads.TaskThread$WrappingRunnable.run(TaskThread.java:61)
	at java.lang.Thread.run(Thread.java:748)
WARN  2018-02-27 10:20:56,069 (org.directwebremoting.util.CommonsLoggingOutput.warn:59) - --Erroring: batchId[4] message[java.lang.AbstractMethodError: Method com/mysql/jdbc/ServerPreparedStatement.setBlob(ILjava/io/InputStream;)V is abstract] `

Here is a screenshot. 
[image]

evranch, 2018-02-27:
OK, I got this working. I had to remove 3.18 and 5.1.7 leaving only mysql-connector-java-5.1.38 in tomcat/lib

I also found then that I could not save a graphical view using the supplied commons-fileupload.jar that is in WEB-INF/lib. When I replaced it with the latest version, 1.3.3, everything seems to be working fine.

The question now is why didn't it work out of the box for me when it seems to have worked for others?

grzesiekb, 2018-02-28:
Would you have time to write test scenarios for the error that you suggest, I tested it and the dedicated environment of the docker on ubuntu and windows works properly. Current version.

grzesiekb, 2018-02-28:
I would need your email to add you to scada-lts.testquality.com

evranch, 2018-03-01:
I probably won't have much time for awhile, I run a ranch and am busy getting ready for lambing time in a couple weeks. I had an emergency where I needed to rapidly develop a system to monitor a cistern and SCADA-LTS is now doing that job, so I'm going to be back outside doing farm work for the next month.

However I will send you my email to add, and if I have some spare time I might try to write a test. Please know that I'm really not a Java guy, I mostly work with embedded C these days so I might not be much help with this project.

closed 2019-10-04 by grzesiekb