SCADA-LTS/Scada-LTS #484: Do we have json api to write tags ? or data points

abhishekricky, 2018-03-01:
Do we have json api to write tags ? or data points

grzesiekb, 2018-03-02:
Yes, We have REST API to save and read data for data point. I will try to prepare an example of the REST API. It is to improve the security of the REST API.

abhishekricky, 2018-03-02:
I just browsed over API, did find following api

"/api/point_value/setValue/{xid}/{type}/{value}"

Value does change on the watchlist point, but it doesnt change on modbus device and later it reverts to original

abhishekricky, 2018-03-02:
@grzesiekb  I tried the SOAP api for writeData but getting below error 

<?xml version="1.0" encoding="UTF-8"?>
<soapenv:Envelope xmlns:soapenv="http://schemas.xmlsoap.org/soap/envelope/" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance">
    <soapenv:Body>
        <soapenv:Fault>
            <faultcode>soapenv:Server.userException</faultcode>
            <faultstring>org.xml.sax.SAXException: No deserializer for {http://www.w3.org/2001/XMLSchema}anyType</faultstring>
            <detail>
                <ns1:hostname xmlns:ns1="http://xml.apache.org/axis/">sularemote</ns1:hostname>
            </detail>
        </soapenv:Fault>
    </soapenv:Body>
</soapenv:Envelope>

grzesiekb, 2018-03-03:
@abhishekricky Could you add the definition of modbus which you didn't update from REST API.

abhishekricky, 2018-03-04:
api end point I used -> /api/point_value/setValue/{xid}/{type}/{value}

/api/point_value/setValue/DP_152354/2/16 

And it was Holding Register -> Address 34 ( Format is unsigned Integer )

grzesiekb, 2018-03-05:
Please write to me email gbylica@softq.pl.

rinose, 2018-11-04:
When I use configureDataSource wsdl soap api I have same issue 
org.xml.sax.SAXException: No deserializer for {http://www.w3.org/2001/XMLSchema}anyType

Have you solved?

Patrykb0802, 2025-01-23:
To edit values of your data points use writeStringData end point. For example:

[image]

vo:itemName = name of your data point
vo:dataType = Java data type not Scada data type, DOUBLE will do in most cases
vo:value = value you want your data point to be set at

For quality and timestamp you can use readData and check what are the parameters of point you want to modify:

[image]

closed 2025-01-23 by Patrykb0802