SCADA-LTS/Scada-LTS #553: Instalation with Docker not working

JoaoCampos89, 2018-04-27:
I am using docker for windows 10. I typed the following commands

`docker pull dockergb/scadalts-dev`

`docker run -it -e DOCKER_HOST_IP=`docker-machine ip` -p 81:8080 dockergb/scadalts-dev /root/start.sh`

The machine is not working, the result printed in the screen:

```
 * Starting MySQL database server mysqld                                              No directory, logging in with HOME=/
                                                                               [fail]
mysql: [Warning] Using a password on the command line interface can be insecure.
ERROR 2002 (HY000): Can't connect to local MySQL server through socket '/var/run/mysqld/mysqld.sock' (2)
/root/start.sh: line 7: ./bin/startup.sh: No such file or directory
java: no process found
/root/start.sh: line 11: cd: /opt/scadalts/webapps/ScadaBR/WEB-INF/classes: Not a directory
sed: can't read env.properties: No such file or directory
/root/start.sh: line 15: ./bin/startup.sh: No such file or directory
tail: cannot open '100' for reading: No such file or directory
tail: cannot open '/opt/scadalts/logs/catalina.out' for reading: No such file or directory
tail: no files remaining

```

katesclau, 2018-04-27:
That is basically saying that MySQL server is not running on the docker container you've just spun up.

by running
`docker container run -it dockergb/scadalts-dev /bin/sh`

you can go inside the container and see the errors MySQL spills at /var/log/mysql/error.log
> [ERROR] Fatal error: Can't open and lock privilege tables: Table storage engine for 'user' doesn't have this option

Which means that MySQL does not have some privileges on key folders. You may add them with
`chown -R mysql:mysql /var/lib/mysql /var/run/mysqld`
or alter the start.sh to do it for you

After solving that, still start script can't find the opt/scadalts/bin/startup.sh ... so I'll keep working on that

katesclau, 2018-04-27:
/opt/scadalts folder inside the docker image is empty, so I'll just have to create it all from scratch

katesclau, 2018-04-30:
Ok, try 

`docker run -it -e DOCKER_HOST_IP=$(docker-machine ip) -p 8080:8080 katesclau/scadalts-dev /tmp/start.sh`

Just had to create from scratch a new one. I'll tag next releases in hub.docker.com for history keeping =D

Also, docker image build repo is here
https://github.com/katesclau/scadalts-docker

Hope this helps

grzesiekb, 2018-05-14:
This problem with line 7 occurs when the docker does not build properly.
Now it is correctly built.
Please:

`docker pull dockergb/scadalts-dev`

`docker run -it -e DOCKER_HOST_IP=`docker-machine ip` -p 81:8080 dockergb/scadalts-dev /root/start.sh`

It should work

closed 2019-10-04 by grzesiekb