SCADA-LTS/Scada-LTS #555: Point Link does not update the value of the target variable

fabiodurao, 2018-04-29:
I installed the version 0.9.4 as taught by the Wiki. I put openjdk8-jre as the main version of Ubuntu 16.04 64bits to keep the project as specified.

I created a point link with an alphanumeric source variable and a binary target variable. I used the following script to convert the values:

if (source.value == "1") {
return true;
} else {
return false;
}

This works in ScadaBR1.1, but in ScadaLTS it only works when I click Save, if I change the value of the source variable nothing happens to the target variable, but if I click Save again the target variable refreshes.

fabiodurao, 2018-04-30:
I managed to get around this problem by creating an event detector in the source variable and creating an event handler with setpoint for the target variable.

viacheg, 2019-08-12:
The same problem. I've found that PointLinkRT always not ready: I think because pointSetComplete() never calls.

mateuszhyski, 2019-08-14:
Hi fabiodurao and viacheg. Is it possible to have some screenshots from our side?

viacheg, 2019-08-17:
Yes, I can make some screenshots and give any other information. But 
what I should capture? May be logs? What I should set in log4j.xml?

14.08.2019 10:46, mateuszhyski пишет:

mateuszhyski, 2019-08-19:
Please , if possible, describe your steps.

mateuszhyski, 2019-10-11:
Solved.

closed 2019-11-18 by mateuszhyski