SCADA-LTS/Scada-LTS #570: Not running on Linux

arpio, 2018-05-11:
Can't get it to run on Linux. ScadaBR from here http://forum.scadabr.com.br/t/scadabr-install-guide-for-a-fresh-install-of-ubuntu-64bit-server-headless-version-16-04-3-using-mariadb-mysql-as-the-database-also-works-on-raspberry-pi-3/1543 is working fine.

But can't get ScadaLTS to work. ScadaLTS is starting and I have the login page but after SignIn nothing happens. I checked the application manager in Tomcat and I cannot start ScadaBR.

I used the second option from the Quick Start tutorial to install ScadaLTS but with Tomcat8 from the Debian repos.

grzesiekb, 2018-05-14:
Yes ScadaLTS is not working on tomcat8

ArkadiuszParafiniuk, 2018-05-14:
Please provide us stakctrace with detais so that we can identify the problem and help you.

Alternatively, you can try to install ScadaLTS using Docker:
https://github.com/SCADA-LTS/Scada-LTS/wiki/Run-ScadaLTS-on-docker---instruction

arpio, 2018-05-18:
It's working now on Debian/Testing:
1. MariaDB from Debian repos.
2. Tomcat 7.0.88 from source
3. ScadaBr.war and ScadaLTS.war from v0.0.9.4 release
4. edit tomcat7/webapps/ScadaBR/WEB-INF/classes/env.properties
db.type=mysql
 #db.datasource=true
#db.datasourceName=java:comp/env/jdbc/scadalts
db.url=jdbc:mysql://localhost:3306/scadalts
db.username=USER
db.password=PWD     
db.pool.maxActive=10
db.pool.maxIdle=10
db.update.log.dir=logs
db.upgrade.check=false

But now the Component List for Views is almost empty. (Only the new UI, everything works in the old UI)

Is it still an installation issue? or is the 0.0.9.4 release just to old?

[image]

closed 2019-10-04 by grzesiekb