SCADA-LTS/Scada-LTS #574: Fix data points import/export

viacheg, 2018-05-15:
Data source type: "MODBUS_IP". 
Export to JSON: dataPoints/dataSourceXid is the same as "xid", not dataSources/xid.
Data point Import is not performed. Result is: "Incorrect result size: expected 1, actual 0".
Repository from "docker pull dockergb/scadalts-dev"

viacheg, 2018-05-16:
I've updated Docker from ver. 18.01 to 18.03 and login works fine

viacheg, 2018-05-16:
Contents of my localhost_access_log.2018-05-16.txt after try to logon:
```
"GET /ScadaLTS/appBody HTTP/1.1" 404 983
"GET /ScadaLTS HTTP/1.1" 302 -
"GET /ScadaLTS/jquery-ui.min.js HTTP/1.1" 404 1001
"GET /ScadaLTS/clipboard.min.js HTTP/1.1" 404 1001
"GET /ScadaLTS/jquery.fancytree-all.js HTTP/1.1" 404 1015
"GET /ScadaLTS/ui.fancytree.min_win7.css HTTP/1.1" 404 1019
"GET /ScadaLTS/ui.fancytree.min.css HTTP/1.1" 404 1009
"GET /ScadaLTS/~@angular/material/core/theming/prebuilt/deeppurple-amber.css HTTP/1.1" 404 1091
"GET /ScadaLTS/~https://fonts.googleapis.com/icon?family=Material+Icons HTTP/1.1" 404 1037
"GET /ScadaLTS/jquery-ui.min.css HTTP/1.1" 404 1003
"GET /ScadaLTS/main.bundle.js HTTP/1.1" 200 288508
"GET /ScadaLTS/jquery-ui.min.js HTTP/1.1" 404 1001
"GET /ScadaLTS/vendor.bundle.js HTTP/1.1" 200 5164500
"GET /ScadaLTS/jquery.fancytree-all.js HTTP/1.1" 404 1015
"GET /ScadaLTS/clipboard.min.js HTTP/1.1" 404 1001
"GET /ScadaLTS/jquery-ui.min.css HTTP/1.1" 404 1003
"GET /ScadaLTS/ui.fancytree.min_win7.css HTTP/1.1" 404 1019
"GET /ScadaLTS/ui.fancytree.min.css HTTP/1.1" 404 1009
"GET /ScadaLTS/~https://fonts.googleapis.com/icon?family=Material+Icons HTTP/1.1" 404 1037
"GET /ScadaLTS/~@angular/material/core/theming/prebuilt/deeppurple-amber.css HTTP/1.1" 404 1091
"GET /ScadaBR/api/auth/admin/admin HTTP/1.1" 404 1007
```

grzesiekb, 2018-05-17:
You can download docker with new version ScadaBR
and
in web browser set http://localhost:81/ScadaBR/api/auth/admin/admin
It should work.

viacheg, 2018-05-17:
I've downloaded new one by `$ ./docker pull dockergb/scadalts-dev` but nothing has changed.
I'm a newbie in docker. How can I be sure that we speak about the same container? I have:
```
$ ./docker inspect e1842b740470
[
    {
        "Id": "sha256:e1842b74047006823593bdca9aa2847fb14053d31b9be4fa881c1069d3a2241b",
        "RepoTags": [
            "dockergb/scadalts-dev:latest"
        ],
        "RepoDigests": [
            "dockergb/scadalts-dev@sha256:882f1568f27aa0c006f8c5b9575de96d68fc019b87ab5cae35edad7548e6378c"
        ],
```

arpio, 2018-05-18:
I also tried the docker installation on Debian (Docker 18.03) just an hour ago. My problem was: "Wrong Username or Password."

The users table in the database is empty. Created an user admin:
"INSERT INTO `users` (`id`, `username`, `password`, `email`, `phone`, `admin`, `disabled`, `lastLogin`, `selectedWatchList`, `homeUrl`, `receiveAlarmEmails`, `receiveOwnAuditEvents`) VALUES (NULL, 'admin', '0DPiKuNIrrVmD8IUCuw1hQxNqZc=', 'admin@yourMangoDomain.com', NULL, 'Y', 'N', NULL, NULL, NULL, '0', 'N');"

Login works but could not create any views, errors when trying to save. Gave up.

JoaoCampos89, 2018-05-18:
I also tried using docker, with that docker/scadalts-dev image, and I having this logs:

mysql: [Warning] Using a password on the command line interface can be insecure.
ERROR 2002 (HY000): Can't connect to local MySQL server through socket '/var/run/mysqld/mysqld.sock' (2)

As docker container can't connect to mysql is not possible to do the login.

closed 2019-10-04 by grzesiekb