SCADA-LTS/Scada-LTS #637: Scada LTS no login

leoboeng, 2018-07-04:
I installed the Scada LTS following the following steps:

Second option:

    Download the latest release.
    Install tomcat7: apt-get install tomcat7.
    Install mysql: apt-get install mysql-server.
    Login to mysql: mysql -u [user] -p.
    Create db:create dababase scadalts;.
    Copy ScadaBR.war & ScadaLTS.war to dir tomacat webapps: cp ScadaBR.war /var/lib/tomcat7/webapps/.
    Restart tomcat7 to generate ScadaLTS folder: /etc/init.d/tomcat7 restart
    Change config file env.properties set connection to db: vim /var/lib/tomcat7/webapps/ScadaBR/WEB-INF/classes/env.properties.
    Restart tomcat7: /etc/init.d/tomcat7 restart

It loads the main page, the login page, but does not enter the login is just loading the page

SoftQ, 2018-07-04:
Hello

Please make sure You set connection to already created database on user 
having full rights to modify it (create/drop table etc.). In case it 
doesn't help:

- please let us know which release have You taken ScadaBR.war from?  We 
recommend the newest one

     https://github.com/SCADA-LTS/Scada-LTS/releases/tag/v0.0.9.4

- provide a screenshot of traffic from web browser "developer 
tools"->Network

- provide us catalina.out log file

  Thank You

  Michał

W dniu 04.07.2018 o 18:08, leoboeng pisze:

leoboeng, 2018-07-04:
Yes I used this link version of ScadaLTS and ScadaBR. Both files are in the folder.

Follow the photos of the errors presented:

[image]

[image]

What catalina.out log file directory?

SoftQ, 2018-07-05:
Hello
As far as Angular (/ScadaLTS) application is concerned, it will not be
developed or supported by us anymore because we lack budget and
competencies for professional Angular  development.  The Angular
(/ScadaLTS) app is also outdated and is not using  new REST API metchods
that we havew implemented on /ScadaBR  application.
Instead /ScadaLTS we will take approach of incremental development basing
on old UI, that is  accessible in browser under /ScadaBR; for this purpose
we will use Vue.js  framework. It will be visible for example as Vue.js
based  point hierarchy Import/Export  functionality in 0.0.9.5 release (it
will be soon available).

Best Regards
Michał

Best REgards

squitel, 2018-07-17:
Hello,
@Michal, do you mean that : 
1) the development of ScadaLTS is stopped and that the efforts will now be put on the improvement of the already existing ScadaBR
or 
2) it is only the development of the new modern Angular GUI (/ScadaLTS) of ScadaLTS that is stopped and that the development of ScadaLTS will go on but with the old GUI (/ScadaBR) ?

As a French native speaker, sorry if it is unclear for me.
The fact that the docker image "dockergb/scadalts" has not been updated during the past 15 months makes me wonder as well...

Thanks in advance.

Best regards,
Jacques

SoftQ, 2018-07-17:
Salut Jacques!

Thank You for contact. Here are  my responses:

1.  Definitively the 2nd option  - we will move our team efforts from
developing Angular UI (/ScadaLTS) only.   ScadaBR UI will instead  be
enriched with Vue.js controls,  1st example will be on hierarchy screen for
import/export functionality. It's at present on testing phase, will be
available in next  release soon. Of course we work a lot in ScadaBR core
towards true ScadaBR 2.0 that shall provide full REST API and significant
usability and  performance improvements comparing to ScadaBR 1.0..

2. Please make sure You follow instruction and "docker pull
dockergb/scadalts-dev".
  Please note the Docker image is build from development branch, therefore
is for testing/evaluation only.  At present we don't provide production
ready Docker image. As far as more "production ready" artifacts are
concerned, please use the newest  ScadaBR.war file from
https://github.com/SCADA-LTS/Scada-LTS/releases section

I have also questions for You:

- Do You have or know persons/companies  that might provide  ideas /
requirements on direction of ScadaBR 2.0 development?
- Can You support or suggest where to look for support for ScadaBR/ScadaLTS
project?  in all terms - work/money/promotion/community
- Do You know on any installations in Europe ? Please write few words in
general.

Merci!
Michał

JoaoCampos89, 2018-07-18:
Hi Michal, I am in Brasil, and I know  persons who use ScadaBR at production. If you want some help you can reach me at my email j.campos893@gmail.com. Some ideas, I think Scada will need support for MQTT, OPC UA, IEC61850 which are protocols needed for  the new industrial of things. MQTT for general purposes in IOT. OPC UA for industrial open technologies and IEC61850 is the protocol used by electrical utilities. Other ideias, might be improve security for 2 factor login and easy deployment.  Relative to promotion and community and I can help also.

squitel, 2018-07-18:
Salut Michal !

Thank you for your answer. 

I've tried the "docker pull dockerdg/scadalts-dev" but couldn't get it working and the symptoms were the same with the latest ScadaBR.war & ScadaLTS.war files as suggested, following these instructions : https://github.com/SCADA-LTS/Scada-LTS/wiki/Install-Scada-LTS-on-linux.
Everything looked ok until I tried to sign in at http://127.0.0.1:8080/ScadaLTS (or from another machine at the corresponding IP) with admin/admin: it never finishes... I remain stucked there forever.
I then tried with http://127.0.0.1:8080/ScadaBR which failed "not available".
I gave it a look in Tomcat manager which listed the /ScadaLTS application but no /ScadaBR.
I therefore tried to upload & deploy from there the ScadaBR.war file but with no success; this reinitialized my session in Tomcat manager...

Note that the non -dev docker image is working flawlessly ("docker pull dockerdg/scadalts").

It's certainly something I did wrong but my several attempts failed as described.
I'm new at Docker but I would have expected a docker image to work out of the box.
I've tried ScadaBR a few years ago (and already found it a pain to install for me) but I finally had it to run.

Regarding your questions:
- I will ask some of my contacts for ideas of future features for ScadaBR and come back to you if something relevant comes out from this.
However, according to me and as Joao wrote above, OPC DA/UA and MQTT would definitely be nice to have!
- Unfortunately my structure is too small to help financially and I don't have the required skills to help develop the product I'm afraid. I was just trying to evaluate ScadaBR/LTS for our needs.
- I don't know any company here in France or in Europe that is using ScadaBR, sorry.

Do you have feedbacks on how ScadaBR can scale in terms of number of "points" (metrics/measures) it can handle?

Best regards,
Jacques

SoftQ, 2018-07-18:
Salut Jacques! Thank You for answers, we have received ‎information on logon problem from two other people and will work to fix  it for sure.- Definitely the /ScadaBR path (application) should be available, without it Scada-LTS - being merely a "front end " is useless and will "not log in"..- Please remember Docker images shall not be deployed on production unless You are able to tune /reconfigure the image -having not only Linux /Java/Tomcat but also deeper container related knowledge . We can help you for free to set up simple evaluation    without Docker which will be surely closer to "production" than Docker.- As far as datapoints number is concerned few years ago in Brazil  there were  few thousands datapoints  on commercial DB and installation is still working / developed.  We support ~8k datapoints installation in Europe with 1-few few second updates /(recomputations (in case of Meta Data Source) using MySQL, however we hope to  ‎to reach soon  ~20k ( still using free and open source DB). To help with bigger installations  we develop now  REST API to automatize adding new datapoints ‎as well screens and components to screens. All in order to help with deployment of few or even dozens of thousand point installations...In case You would like to evaluate,  as I have written, we can help You to set stable running environment  for free or very small fee depending what you want. In case of evaluation  success we can surely come to agreement on  some payment for 2nd part of our work which might  consist some deployment automation,  tuning or small  hardening.  For very professional deployment, support in France area  there is for sure to be recommended AbilIT company, which is the main leader and sponsor of the project.                                                                                                                                     All the best! Good luck with Scada-LTS ‎! In case You need any help, please don't hesitate to write directly to <scada-lts@softq.pl> email.Thanks                                                                                                                                                                                                    Michał                                                                                                                                                                                                                Od: squitelWysłano: środa, 18 lipca 2018 19:58Do: SCADA-LTS/Scada-LTSOdpowiedz: SCADA-LTS/Scada-LTSDW: SoftQ; CommentTemat: Re: [SCADA-LTS/Scada-LTS] Scada LTS no login (#637)Salut Michal !
Thank you for your answer.
I've tried the "docker pull dockerdg/scadalts-dev" but couldn't get it working and the symptoms were the same with the latest ScadaBR.war & ScadaLTS.war files as suggested, following these instructions : https://github.com/SCADA-LTS/Scada-LTS/wiki/Install-Scada-LTS-on-linux.
Everything looked ok until I tried to sign in at http://127.0.0.1:8080/ScadaLTS (or from another machine at the corresponding IP) with admin/admin: it never finishes... I remain stucked there forever.
I then tried with http://127.0.0.1:8080/ScadaBR which failed "not available".
I gave it a look in Tomcat manager which listed the /ScadaLTS application but no /ScadaBR.
I therefore tried to upload & deploy from there the ScadaBR.war file but with no success; this reinitialized my session in Tomcat manager...
Note that the non -dev docker image is working flawlessly ("docker pull dockerdg/scadalts").
It's certainly something I did wrong but my several attempts failed as described.
I'm new at Docker but I would have expected a docker image to work out of the box.
I've tried ScadaBR a few years ago (and already found it a pain to install for me) but I finally had it to run.
Regarding your questions:

I will ask some of my contacts for ideas of future features for ScadaBR and come back to you if something relevant comes out from this.
However, according to me and as Joao wrote above, OPC DA/UA and MQTT would definitely be nice to have!
Unfortunately my structure is too small to help financially and I don't have the required skills to help develop the product I'm afraid. I was just trying to evaluate ScadaBR/LTS for our needs.
I don't know any company here in France or in Europe that is using ScadaBR, sorry.

Do you have feedbacks on how ScadaBR can scale in terms of number of "points" (metrics/measures) it can handle?
Best regards,
Jacques

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/SCADA-LTS/Scada-LTS/issues/637#issuecomment-406020432",
"url": "https://github.com/SCADA-LTS/Scada-LTS/issues/637#issuecomment-406020432",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
},
{
"@type": "MessageCard",
"@context": "http://schema.org/extensions",
"hideOriginalBody": "false",
"originator": "AF6C5A86-E920-430C-9C59-A73278B5EFEB",
"title": "Re: [SCADA-LTS/Scada-LTS] Scada LTS no login (#637)",
"sections": [
{
"text": "",
"activityTitle": "**squitel**",
"activityImage": "https://assets-cdn.github.com/images/email/message_cards/avatar.png",
"activitySubtitle": "@squitel",
"facts": [

]
}
],
{
"name": "Add a comment",
"@type": "ActionCard",
"inputs": [
{
"isMultiLine": true,
"@type": "TextInput",
"id": "IssueComment",
"isRequired": false
}
],
"actions": [
{
"name": "Comment",
"@type": "HttpPOST",
[... cut ...]
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueClose\",\n\"repositoryFullName\": \"SCADA-LTS/Scada-LTS\",\n\"issueId\": 637\n}"
},
{
"targets": [
{
"os": "default",
"uri": "https://github.com/SCADA-LTS/Scada-LTS/issues/637#issuecomment-406020432"
}
],
"@type": "OpenUri",
"name": "View on GitHub"
},
{
"name": "Unsubscribe",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"MuteNotification\",\n\"threadId\": 353058134\n}"
}
],
"themeColor": "26292E"
}
]

squitel, 2018-07-19:
Salut Michal,
Thanks for these additional pieces of information.
I'm also grateful to you for your kind offer for help. However I'll keep on putting some more efforts on my side before, potentially, coming back to you.
Thanks again and let's keep in touch.
Best regards,
Jacques

closed 2019-10-04 by grzesiekb