SCADA-LTS/Scada-LTS #723: Periodic purge data time configurable

danibeni, 2018-10-16:
For a medium size database, the data purge can take some time and the SCADA performance is decreased during that time. The purge is fixed in code at 03:05 hours, and I need to SCADA to be stable during all night, so I think it wolud be a good idea to extract that fixed time to a system settings parameter. I have already done it, so I can create a branch to include this feature if you considered it interesting

SoftQ, 2018-10-19:
Thank You for information!  Please feel free to make pull request to "develop" branch with Your changes.  We will review it and - hopefully - merge.

SoftQ, 2018-10-23:
> I have the code ready to make pull request, but I do not find the way to
> do it without being collaborator. Sorry for the silly question, but could
> someone tell me how to make the pull request?

Hello, Please:
-  make sure You have forked from develop branch
-  follow this guide
https://help.github.com/articles/creating-a-pull-request-from-a-fork/

Thank You!

SoftQ, 2018-11-07:
PS. your pull request has been approved. Please feel free to merge it !