SCADA-LTS/Scada-LTS #729: Documentation: Navigation through graphic views

pablorq, 2018-10-24:
Hello,

There's any documentation on how to navigate through graphic views, without selecting the view from the drop down menu?

I would like to have the graphic view fullscreen and move form one view to another.

Regards.

SoftQ, 2018-10-26:
Hello. Navigating between views without using dropdown is possible to be done at least in two ways - by  adding to given view

- **Link** component; there You choose which view it shall direct to)
- **HTML** component; there You have to write and style Your custom HTML that shall include a link - every view has it's own link.
 
Good luck!

grzesiekb, 2019-10-07:
I am thinking about a dedicated navigation component

closed 2019-10-07 by grzesiekb