SCADA-LTS/Scada-LTS #731: Error while enabling or disabling data source via script

hilariodaufemback, 2018-10-25:
Ao executar um script com o seguinte codigo: 
ds.disableDataSource("DS_587499");

o mesmo retorna:
Error org.mozilla.javascript.WrappedException: Wrapped com.serotonin.mango.vo.permission.PermissionException: Not logged in (#1) at org.mozilla.javascript.Context.throwAsScriptRuntimeEx(Context.java:1754) at org.mozilla.javascript.MemberBox.invoke(MemberBox.java:148) at org.mozilla.javascript.NativeJavaMethod.call(NativeJavaMethod.java:225) at org.mozilla.javascript.Interpreter.interpretLoop(Interpreter.java:1473) at script.__scriptExecutor__(:1) at script(:4) at org.mozilla.javascript.Interpreter.interpret(Interpreter.java:815) at org.mozilla.javascript.InterpretedFunction.call(InterpretedFunction.java:109) at org.mozilla.javascript.ContextFactory.doTopCall(ContextFactory.java:394) at org.mozilla.javascript.ScriptRuntime.doTopCall(ScriptRuntime.java:3091) at ................

Li nos blogs do ScadaBR que isso foi detectado com um problema e iria ser ajustado. Esse ajuste veio para o ScadaLTS? Se não fica a solicitação de ajustes. Obrigado

SoftQ, 2018-10-26:
Thank You for information, we confirm there is issue with disabling datasource from Javascript level. We will investigate it. It would pe perfect if  You can send link to forum or  - better - indicate part of code that contains fix for this issue.

SoftQ, 2018-11-13:
We have corrected the error, however we haven't found where (and if) it is corrected in ScadaBR (have looked at newest code available  - 1.1-RC).
Therefore our solution requires testing in Your case; surely it works in simple cases however it may occur some scenarios are not covered by it  (i.e. when Common.getUser() returns null )

Xt-Man, 2022-09-28:
Hi, i'm running ScadaLTS docker image v2.7.1.1. Trying to run ds.enableDataSource('DS_347815') and i'm receiving the same error message, any ideas? Thanks

Limraj, 2022-09-28:
Hello,
docker image with fix: pr-2341

Limraj, 2022-09-29:
@Xt-Man Let me know if it's okay.

closed 2022-09-29 by Limraj