SCADA-LTS/Scada-LTS #747: Docker install doesn't work - Tomcat ok, ScadaBR not running

pablorq, 2018-11-14:
Hi,

A new docker image install doesn't work correctly. Tomcat is running but ScadaBR is not.

DETAILS

In a Windows 7 64bit PC with Docker installed and checked is ok (hello_world is ok), run the following commands:
docker-machine start
docker run -it -e DOCKER_HOST_IP=192.168.99.100 -p 81:8080 scadalts/scadalts /root/start.sh

The docker image is downloaded and started.

Open a browser (Firefox 63.0.1 - 64bit) and try the following addresses:
URL: http://192.168.99.100:81
RESPONSE: Tomcat starting page

URL: http://192.168.99.100:81/ScadaBR/
RESPONSE;
Estado HTTP 404 - /ScadaBR/
type Informe de estado
mensaje /ScadaBR/
descripción El recurso requerido no está disponible.
Apache Tomcat/7.0.85
(Tomcat error page)

More info from 'docker images --digests'
DIGEST: sha256:2974fd40679035e659754a130fe9328e10f3626449835e5e1231c3959309bd7c   
IMAGE ID: 73190fdc87ee

Any idea?

Regards,

pablorq, 2018-11-14:
Hi,

Trying with docker 'develop' image updated 1h ago, but same issue.

:(

Any idea?

pablorq, 2018-11-15:
RAM: 16GB 

Last week I was working with a previous version/tag of scadalts.

Working on Docker management I remove the docker image. Then I ran:
docker run -it -e DOCKER_HOST_IP=192.168.99.100 -p 81:8080 scadalts/scadalts /root/start.sh

... and a new docker image was downloaded. Same PC.

But this new image have this behavior: docker container works, Tomcat is running and ScadaBR shows a tomcat error page as a broken link.

Would you like I run some test on the docker container?

Regards.

pablorq, 2018-11-29:
Hi,

After some tests I delete the container and run it again. Now it works ok.

No idea about what was the problem.

Closed because I can't reproduce the issue.

Regards.

closed 2018-11-29 by pablorq

SoftQ, 2018-11-30:
Thank You, we also could not reproduce this issue!