SCADA-LTS/Scada-LTS #770: Data points for MBUS datasource cannot be added

slsimic, 2018-12-20:
After adding MBUS data source, setting IP address, port and search range for primary address,
and when search is done, I get MBUS device recognized, 
having all of the readings correct on the expanded tree.
But when I try to add a point by clicking "add" icon on the right side of the values, nothing happens.
No data points are added in the list below.
I've noticed that after I click on the small "add" icon, in the localhost_access_log the line with an error 
"POST /ScadaLTS/dwr/call/plaincall/**MiscDwr.jsError.dwr** HTTP/1.1" 200 89

> "POST/ScadaLTS/dwr/call/plaincall/MiscDwr.doLongPoll.dwr HTTP/1.1" 200 88
> "GET/ScadaLTS/data_sources.shtm HTTP/1.1" 200 43279
> "POST/ScadaLTS/dwr/call/plaincall/MiscDwr.terminateLongPoll.dwr HTTP/1.1" 200 90
> "POST/ScadaLTS/dwr/call/plaincall/MiscDwr.doLongPoll.dwr HTTP/1.1" 200 103
> "GET/ScadaLTS/dwr/interface/DataSourceListDwr.js HTTP/1.1" 200 3062
> "GET/ScadaLTS/dwr/interface/MiscDwr.js HTTP/1.1" 200 4379
> "POST/ScadaLTS/dwr/call/plaincall/DataSourceListDwr.init.dwr HTTP/1.1" 200
> "GET/ScadaLTS/data_source_edit.shtm?dsid=3 HTTP/1.1" 200 96236
> "POST/ScadaLTS/dwr/call/plaincall/MiscDwr.terminateLongPoll.dwr HTTP/1.1" 200 89
> "POST/ScadaLTS/dwr/call/plaincall/MiscDwr.initializeLongPoll.dwr HTTP/1.1" 200 102
> "GET/ScadaLTS/dwr/interface/MiscDwr.js HTTP/1.1" 200 4379
> "GET/ScadaLTS/dwr/interface/DataSourceEditDwr.js HTTP/1.1" 200 30026
> "POST/ScadaLTS/dwr/call/plaincall/DataSourceEditDwr.editInit.dwr HTTP/1.1" 200 198
> "POST/ScadaLTS/dwr/call/plaincall/DataSourceEditDwr.searchMBus.dwr HTTP/1.1" 200 89
> "POST/ScadaLTS/dwr/call/plaincall/DataSourceEditDwr.mBusSearchUpdate.dwr HTTP/1.1"
> "POST/ScadaLTS/dwr/call/plaincall/DataSourceEditDwr.mBusSearchUpdate.dwr HTTP/1.1"
> "POST/ScadaLTS/dwr/call/plaincall/DataSourceEditDwr.mBusSearchUpdate.dwr HTTP/1.1"
> "POST/ScadaLTS/dwr/call/plaincall/DataSourceEditDwr.mBusSearchUpdate.dwr HTTP/1.1"
> "POST/ScadaLTS/dwr/call/plaincall/DataSourceEditDwr.getMBusResponseFrames.dwr
> "POST /ScadaLTS/dwr/call/plaincall/MiscDwr.initializeLongPoll.dwr HTTP/1.1" 200 87
> "POST /ScadaLTS/dwr/call/plaincall/MiscDwr.jsError.dwr HTTP/1.1"
> "POST/ScadaLTS/dwr/call/plaincall/MiscDwr.doLongPoll.dwr HTTP/1.1" 200 87

If there is  another log file to look into to help resolve the problem, please let me know.

grzesiekb, 2019-03-07:
We do not maintain the data source MBUS you can try to use node-red to download data and save to file and host and use HttpRetriver to read this data.

slsimic, 2024-04-04:
> We do not maintain the data source MBUS you can try to use node-red to download data and save to file and host and use HttpRetriver to read this data.

Who does maintain or even know something about MBUS data sources in the Scada-LTS?
I tried newest release 2.6.18 with Linux installer within its own java/mysql/tomcat environment and I must say, situation is even worse than more than 5 years ago. Now I cannot get to adding the data points because when Search is clicked at the MBUS data source, nothing is found and error in Tomcat log regarding the Scada-LTS repeats while different MBUS addresses are scanned

ERROR 2024-04-04T14:14:33,962 (org.directwebremoting.util.CommonsLoggingOutput.error:75) - No converter found for 'net.sf.mbus4j.MBusAddressing' 

Now I know that this has something to do with maven's java mbus libraries but searching 'no converter' or 'MBusAddressing' related to net.sf.mbus4j libraries returns ZERO results on Google search.