SCADA-LTS/Scada-LTS #785: Only fullscreen mode on login for readonly user, and/or hide features from readonly user

djwho, 2019-01-16:
Hi, would be usefull to have a feature to hide options that are not used by users, even only a "fullscreen-only" with default page/view. Any chance for that? Cheers.

jantoniaze, 2019-01-20:
Bom dia! Muito útil esta opção seria definição de usuário como desenvolvedor grupo ou perfil e outro grupo para operação, para usuário final e muito interessante.

grzesiekb, 2019-01-25:
@jantoniaze  Please write in English.

grzesiekb, 2019-01-25:
The idea is cool

djwho, 2019-01-25:
Yes, would be great to have two (as i understand) groups of users:
-scada developers/maintainers and
-scada users - with "feature hiding" ability 
If needed i can help with testing.
Maybe a third option next to "users" and "user profiles" a "viewing profile" that we can also connect to user.

SoftQ, 2019-01-25:
There are many useful improvements that might  be implemented around user rights:- restricting access (hiding) menu entries like watchlist, datasources etc.‎ It should be relatively easy to implement it - restricting access on views only to selected view components/"layers" (buttons, etc.)‎ Such functionality surely wilrequire more analysis/ design and implementation effort.‎With regard to fullscreen only access for "readonly" ~users at present it might be achieved by defining  anonymous views + setting additional authentication+authorization on Tomcat level. In case You would like to have some assistance,  please let us know  to scada-lts@softq.pl                                                                                                                                      Best Regards                                                                                                                                                                                                    Michał‎ Rokitiański

djwho, 2019-01-25:
@SoftQ by "viewing profile" i meant interface view - report button, watch list. Restrict access to created scada views i think can be achieved by making 2 separate views for 2 users just for now (did you meant that?)
A "fullscreen only" feature i use by now as a workaround to not show anything else but one scada view. But this can be also additional feature. BTW ScadaLTS is a nice peace of software ;-)

jantoniaze, 2019-01-26:
Good evening, I am very happy to know that this group of developers are with this project at full steam, I use scadalts in the factory that we work, great plan for this supervisor, where our project and put the factory in the palm of our director, maintainers, operators. I am grateful for liking the idea because it would help a lot by creating these development and operation profiles with full-screen option for a button in the development of the graphic representation. or something like that. They are just suggestions for scadalts lovers friends.

jantoniaze, 2019-05-25:
Good morning, we have some forecast of the next release of scadalts.

grzesiekb, 2019-10-07:
Yes, now we are planning tasks: https://github.com/SCADA-LTS/Scada-LTS/milestones

closed 2019-10-07 by grzesiekb