SCADA-LTS/Scada-LTS #835: Language switch to Chinese, the page appears unreadable code

iotclouddeveloper, 2019-04-09:
Hello,
I'm going to run the scadabr.war (v.0.0.9.8),After I login in, I switch the language to Chinese, the page appears unreadable code.
where is the problem???
thanks.

grzesiekb, 2019-05-27:
I do not have that effect.
Do you use the scada version on the [docker](https://hub.docker.com/r/scadalts/scadalts)?

iotclouddeveloper, 2019-05-30:
[image]
I donot use the scada version on the docker ,I deploy v.0.0.9.8 on the tomcat.
I found  that there are less items in messages_zh.properties than in messages_en.properties. Is that the reason?

liaoah, 2019-08-30:
@grzesiekb 
the docker also have the same problem

grzesiekb, 2019-10-07:
@liaoah Please print screen from docker.

Limraj, 2023-06-03:
Hi @iotclouddeveloper and @liaoah,
We fixed it in this issue: https://github.com/SCADA-LTS/Scada-LTS/issues/2421

The problem with encoding and setting the language for the user should not occur from version 2.7.5.3, (2.7.4.1 fixed encoding, and 2.7.5.3 added ability save lang per user)

Regards,
Kamil Jarmusik

closed 2023-06-03 by Limraj