SCADA-LTS/Scada-LTS #837: centos7.4 Scada LTS no login of web 404

he77pc, 2019-04-20:
I installed the Scada LTS following the following steps:

pache Tomcat/7.0.94 |
 java 1.8.0_201-b09 | 
centos7.4  | 3.10.0-957.10.1.el7.x86_64
-- 
mysql of mariadb
mariadb-java-client-2.4.1.jar
/var/lib/tomcat7/lib  of  
/var/lib/tomcat7/webapps/ScadaBR/WEB-INF/lib

env.properties
-- 
db.type=mysql
db.datasource=true
db.datasourceName=java:comp/env/jdbc/scadalts
db.url=jdbc:mariadb://localhost:3306/scadalts
db.username=root
db.password=xxxxxx
db.pool.maxActive=10
db.pool.maxIdle=10
db.update.log.dir=logs
db.upgrade.check=false
-- 
Tomcat Web Application Manager

/ScadaBR | None specified |   | false
-- | -- | -- | --

he77pc, 2019-04-20:
Exception sending context initialized event to listener instance of class com.serotonin.mango.MangoContextListener
java.lang.IllegalArgumentException: No DataSource specified
	at org.springframework.util.Assert.notNull(Assert.java:134)
	at org.springframework.jdbc.datasource.DataSourceUtils.doGetConnection(DataSourceUtils.java:97)
	at org.springframework.jdbc.datasource.DataSourceUtils.getConnection(DataSourceUtils.java:77)
	at org.springframework.jdbc.core.JdbcTemplate.execute(JdbcTemplate.java:382)
	at org.springframework.jdbc.core.JdbcTemplate.execute(JdbcTemplate.java:428)
	at com.serotonin.mango.db.MySQLAccess.newDatabaseCheck(MySQLAccess.java:105)
	at com.serotonin.mango.db.DatabaseAccess.initialize(DatabaseAccess.java:118)
	at com.serotonin.mango.MangoContextListener.databaseInitialize(MangoContextListener.java:417)
	at com.serotonin.mango.MangoContextListener.contextInitialized(MangoContextListener.java:105)
	at org.apache.catalina.core.StandardContext.listenerStart(StandardContext.java:5197)
	at org.apache.catalina.core.StandardContext.startInternal(StandardContext.java:5720)
	at org.apache.catalina.util.LifecycleBase.start(LifecycleBase.java:145)
	at org.apache.catalina.core.ContainerBase.addChildInternal(ContainerBase.java:1018)
	at org.apache.catalina.core.ContainerBase.addChild(ContainerBase.java:994)
	at org.apache.catalina.core.StandardHost.addChild(StandardHost.java:652)
	at org.apache.catalina.startup.HostConfig.deployWAR(HostConfig.java:1127)
	at org.apache.catalina.startup.HostConfig$DeployWar.run(HostConfig.java:2020)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)

SoftQ, 2019-04-20:
Please refer to Docker container configuration.

 We use to configure datasource on Tomcat level ( xml configuration  file) . 
For Tomcat level datasource configuration  please  put database jar (for  mariadb connection) inside tomcat/lib folder.
‎

closed 2019-05-27 by grzesiekb