SCADA-LTS/Scada-LTS #908: Docker - New install - Cannot find user Id

pablorq, 2019-06-28:
Hi,

In a new installation with Docker on Windows 7 using Docker Toolbox, ScadaLTS shows the login and after admin:admin it returns "Cannot find user Id".

Steps to reproduce this issue:
- Remove the docker machine
docker-machine rm default
- Create a new machine
docker-machine create --driver virtualbox default
- Get host ip
docker-machine ip
192.168.99.10
- Create container
docker run -it -e DOCKER_HOST_IP=192.168.99.100 -p 81:8080 scadalts/scadalts /root/start.sh

Result:
- 192.168.99.10:81
Tomcat homepage
- 192.168.99.10:81/ScadaBR/
Login page
- Enter user:pass
user: admin
pass: admin
- Hit 'Login'
-> Cannot find user Id

I've tested several times and it doesn't work.

Any idea?

Regards

SoftQ, 2019-06-28:
It is possible the database has not ptoperly initialized while Starting Docker container the first time. Try to relaese RAM memory on PC, play with "--memory-reservation=..." docker option  or simply try on other PC.

pablorq, 2019-06-28:
Hi!

Last week it works well in this PC. Then something went wrong when installing other containers (nextcloud and others).

Because of that ScadaLTS stops to work (This is for another ticket). As a last solution I re-created the all machine.

The PC has a lot of memory an processors available for docker. I don't think this is the problem.

Regards...

pablorq, 2019-06-28:
Hi,

I found something: There are no users on the users table, neither the admin user.

```
mysql> show databases;
+--------------------+
| Database           |
+--------------------+
| information_schema |
| mysql              |
| performance_schema |
| scadalts           |
| sys                |
+--------------------+
5 rows in set (0.01 sec)

mysql> use scadalts;
Reading table information for completion of table and column names
You can turn off this feature to get a quicker startup with -A

Database changed
mysql> show table;
ERROR 1064 (42000): You have an error in your SQL syntax; check the manual that corresponds to your MySQL server version for the right syntax to use near '' at line 1
mysql> show tables;
+------------------------+
| Tables_in_scadalts     |
+------------------------+
| compoundEventDetectors |
| dataPointUsers         |
| dataPoints             |
| dataSourceUsers        |
| dataSources            |
| eventHandlers          |
| events                 |
| flexProjects           |
| mailingListInactive    |
| mailingListMembers     |
| mailingLists           |
| mangoViewUsers         |
| mangoViews             |
| pointEventDetectors    |
| pointHierarchy         |
| pointValueAnnotations  |
| pointValues            |
| reportInstanceData     |
| reportInstancePoints   |
| reportInstances        |
| reports                |
| scheduledEvents        |
| schema_version         |
| scripts                |
| systemSettings         |
| userComments           |
| userEvents             |
| users                  |
| watchListPoints        |
| watchListUsers         |
| watchLists             |
+------------------------+
31 rows in set (0.00 sec)

mysql> select * from users;
Empty set (0.00 sec)
```

From another ticket (https://github.com/SCADA-LTS/Scada-LTS/issues/574#issuecomment-390085315) I found a similar problem with the missing admin user.

After adding the missing admin user:
```
mysql> INSERT INTO users (id, username, password, email, phone, admin, disabled, lastLogin, selectedWatchList, homeUrl, receiveAlarmEmails, receiveOwnAuditEvents) VALUES (NULL, 'admin', '0DPiKuNIrrVmD8IUCuw1hQxNqZc=', 'admin@yourMangoDomain.com', NULL, 'Y', 'N', NULL, NULL, NULL, '0', 'N');
Query OK, 1 row affected (0.06 sec)

mysql> select * from users;
+----+----------+------------------------------+---------------------------+-------+-------+----------+-----------+-------------------+---------+--------------------+-----------------------+
| id | username | password                     | email                     | phone | admin | disabled | lastLogin | selectedWatchList | homeUrl | receiveAlarmEmails | receiveOwnAuditEvents |
+----+----------+------------------------------+---------------------------+-------+-------+----------+-----------+-------------------+---------+--------------------+-----------------------+
|  1 | admin    | 0DPiKuNIrrVmD8IUCuw1hQxNqZc= | admin@yourMangoDomain.com | NULL  | Y     | N        |      NULL |              NULL | NULL    |                  0 | N                     |
+----+----------+------------------------------+---------------------------+-------+-------+----------+-----------+-------------------+---------+--------------------+-----------------------+
1 row in set (0.00 sec)
```

Now I'm able to login.

But something is still broken. If I try to enter in the 'Users' menu, I get the following error:

> StatementCallback; bad SQL grammar [select u.id, u.name, u.xid from usersProfiles u order by u.name]; nested exception is com.mysql.jdbc.exceptions.jdbc4.MySQLSyntaxErrorException: Table 'scadalts.usersProfiles' doesn't exist

Any idea of how to re-create a initial working database?

By the way: something is broken in the last docker image about the database.

Regards...

pablorq, 2019-07-02:
Hi,

I was doing some research about this issue.

SYSTEM USED: Windows 7 + Docker Toolbox

**SUMMARY 1: Port forwarding must be done after create the container.**
**SUMMARY 2: GIT Bash and CMD has different behaviors.**

#GENERAL TESTS CONDITIONS
The tests were done on the same pc with the same installation of Docker Toolbox and VirtualBox. During the tests the docker machine is removed and created again, and containers are created inside the docker machine.

The scadalts image used is the last available:

```
Id:sha256:4a8578395dbe566a3d744c85b6c01b880121e40e93bc9ede88ed01c8b4557791
Created:2019-06-04T12:30:05.820328836Z
```

The scadalts image was downloaded and saved locally, to use always the same image:

```
docker save -o scadalts.tar scadalts/scadalts
```

All the code specified before was run on the CMD windows command terminal. All the code run on GIT Bash will be specified.

#EXPECTED BEHAVIOR
The ScadaLTS application will be installed on a windows 7 pc and accessed by other pcs on the same network. For this reason is necessary to forward the docker container port trough the docker machine (the virtualbox machine).

This is achieved by adding a port forward at the virtualbox virtual machine, using the following command:

```
"C:\Program Files\Oracle\VirtualBox\vboxmanage.exe" modifyvm default --natpf1 "scada,tcp,,81,,81"
```

Assuming that the docker machine has the IP 192.168.99.100, is expected to reach the scadalts app from the following URLs:
local - 192.168.99.100:81/ScadaBR/
local - localhost:81/ScadaBR/
remote - [pc network ip]:81/ScadaBR/

#GENERATE THE ERROR
A simple way to generate the error is to add the port forwarding before create the container:

```
docker-machine stop
docker-machine rm -y default
docker-machine create --driver virtualbox default
docker load -i "scadalts.tar"
docker-machine stop
"C:\Program Files\Oracle\VirtualBox\vboxmanage.exe" modifyvm default --natpf1 "scada,tcp,,81,,81"
docker-machine start
docker run -e DOCKER_HOST_IP=192.168.99.100 -p 81:8080 --name scada scadalts/scadalts /root/start.sh
```

Trying to access the app (192.168.99.100:81/ScadaBR/) shows that Tomcat is running but the ScadaBR webapp isn't working:

```
Status HTTP 404 - /ScadaBR/login.htm
type Status report
message /ScadaBR/login.htm
description The requested resource is not available.
Apache Tomcat/7.0.85
```

#SAME STEPS ON GIT BASH SHOW A DIFFERENT ERROR
Running the same steps on GIT Bash provides a different error:

```
docker-machine stop
docker-machine rm -y default
docker-machine create --driver virtualbox default
docker load -i "scadalts.tar"
docker-machine stop
"C:\Program Files\Oracle\VirtualBox\vboxmanage.exe" modifyvm default --natpf1 "scada,tcp,,81,,81"
docker-machine start
docker run -e DOCKER_HOST_IP=192.168.99.100 -p 81:8080 --name scada scadalts/scadalts /root/start.sh
```

When you access the app, now ScadaBR webapp seems to work, but after login the ' Cannot find user Id' error is shown. Looking on the DB inside the container, the DB was created partially and some tables are missing.

#NO PORT FORWARD
The same steps without port forwarding seems to work well:

```
docker-machine stop
docker-machine rm -y default
docker-machine create --driver virtualbox default
docker load -i "scadalts.tar"
docker run -e DOCKER_HOST_IP=192.168.99.100 -p 81:8080 --name scada scadalts/scadalts /root/start.sh
```

The ScadaBR webapp works ok.

#PORT FORWARD AFTER CONTAINER CREATION
Now the steps are changed creating the container before adding the port forwarding to VirtualBox:

```
docker-machine stop
docker-machine rm -y default
docker-machine create --driver virtualbox default
docker load -i "scadalts.tar"
docker run -e DOCKER_HOST_IP=192.168.99.100 -p 81:8080 --name scada scadalts/scadalts /root/start.sh
docker-machine stop
"C:\Program Files\Oracle\VirtualBox\vboxmanage.exe" modifyvm default --natpf1 "scada,tcp,,81,,81"
docker-machine start
docker start scada
```

The ScadaBR webapp works ok.

#ANY PORT FORWARD BREAKS THE APP
If you add any other port forwarding seems to also break the app:

```
docker-machine stop
docker-machine rm -y default
docker-machine create --driver virtualbox default
docker load -i "scadalts.tar"
docker-machine stop
"C:\Program Files\Oracle\VirtualBox\vboxmanage.exe" modifyvm default --natpf1 "scada,tcp,,8100,,8100"
docker-machine start
docker run -e DOCKER_HOST_IP=192.168.99.100 -p 81:8080 --name scada scadalts/scadalts /root/start.sh
```

Trying to access the app (192.168.99.100:81/ScadaBR/) shows that Tomcat is running but the ScadaBR webapp isn't working:

```
Status HTTP 404 - /ScadaBR/login.htm
type Status report
messaje /ScadaBR/login.htm
descripción The requested resource is not available.
Apache Tomcat/7.0.85
```

Have you any idea why the docker scadalts is behaving this way?

Limraj, 2023-06-03:
Hi @pablorq,
Most likely database migration failed, no admin user is a symptom if the database is running.

All information about migrations can be found in the table:
scadalts.schema_version .
Now if you want to repeat a failed migration, you should remove the entry with success=0 and restart tomcat with the application.
If you want a given migration step to be skipped, for a row with a failed migration, you should change the success flag from 0 to 1, although it is usually not recommended.

Regards,
Kamil Jarmusik

closed 2023-06-03 by Limraj