SCADA-LTS/Scada-LTS #940: repairing the docker

grzesiekb, 2019-07-15:


closed 2019-07-15 by grzesiekb

pablorq, 2019-08-05:
Hi,

This change is already updated on docker?

The docker image available today is from 04/06/19.

Regards.

grzesiekb, 2019-08-05:
@pablorq  this is for checking
(21 days ago marge to new/develop) create docker:new_develop (tag: new_develop)
but last change docker:new_develop is 20 days ago 
[image]

pablorq, 2020-04-02:
Sorry for the late response.

As you show in the picture, the changes are on the 'new_develop' branch and no in the 'latest'.

My question was about to provide the changes on the latest docker release.

Thank you anyway!

grzesiekb, 2020-04-02:
Please use the scadals/scadalts:siv_dev.
I will try to make sure that after installation on the production of the last version, it will be released on docker scadalts/scadalts:latest

pablorq, 2020-04-08:
Ok. I will try it in the future.

Thank you!