SCADA-LTS/Scada-LTS #949: Bad special characters

jeffersonpimenta, 2019-07-19:
After install and switch to portuguese language, i got bad characters, example:

UsuÃ¡rio   instead of Usuário
AtÃ©       instead of Até
GrÃ¡fico   instead of Gráfico
prÃ³prio  instead of próprio
ExceÃ§Ã£o  instead of Exceção
PerÃ­odo     instead of período
atualizaÃ§Ã£o   instead of atualização

and so on...

jeffersonpimenta, 2019-08-06:
Partial solution:

Update /var/lib/tomcat7/conf/*.xml  to utf-8

But i still have "Informa��o" on the top center.

grzesiekb, 2019-08-07:
@jeffersonpimenta  It works correctly on the docker

[image]

jeffersonpimenta, 2019-08-07:
How can i contribute for translations and corrections?

Example: in modbus serial datasource i have "Quantizção" instead of "Quantização", missing a "a" letter

jeffersonpimenta, 2019-08-07:
On graphical views, enhanced image chart, i got this

[image]

Limraj, 2023-04-17:
Discussion: https://github.com/SCADA-LTS/Scada-LTS/discussions/2420 (issue #2421 and #2370)
Is corrected from [2.7.4.1](https://github.com/SCADA-LTS/Scada-LTS/releases/tag/v2.7.4.1) Scada-LTS

closed 2023-04-17 by Limraj