SCADA-LTS/Scada-LTS #952: ScadaLTS on RPI

jeffersonpimenta, 2019-07-24:
After try to run the new version (0.9.9) of ScadaLTS on RPI i got an issue, using tomcat7.0.94 and java 8

My log file says:

Jul 24, 2019 5:09:44 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: Server version:        Apache Tomcat/7.0.94
Jul 24, 2019 5:09:44 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: Server built:          Apr 10 2019 16:56:40 UTC
Jul 24, 2019 5:09:44 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: Server number:         7.0.94.0
Jul 24, 2019 5:09:44 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: OS Name:               Linux
Jul 24, 2019 5:09:44 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: OS Version:            4.15.0-1031-raspi2
Jul 24, 2019 5:09:44 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: Architecture:          aarch64
Jul 24, 2019 5:09:44 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: Java Home:             /usr/lib/jvm/java-8-openjdk-arm64/jre
Jul 24, 2019 5:09:44 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: JVM Version:           1.8.0_212-8u212-b03-0ubuntu1.18.04.1-b03
Jul 24, 2019 5:09:44 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: JVM Vendor:            Oracle Corporation
Jul 24, 2019 5:09:44 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: CATALINA_BASE:         /opt/tomcat7
Jul 24, 2019 5:09:44 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: CATALINA_HOME:         /opt/tomcat7
Jul 24, 2019 5:09:44 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: Command line argument: -Djava.util.logging.config.file=/opt/tomcat7/conf/logging.properties
Jul 24, 2019 5:09:44 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: Command line argument: -Djava.util.logging.manager=org.apache.juli.ClassLoaderLogManager
Jul 24, 2019 5:09:44 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: Command line argument: -Djdk.tls.ephemeralDHKeySize=2048
Jul 24, 2019 5:09:44 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: Command line argument: -Dignore.endorsed.dirs=
Jul 24, 2019 5:09:44 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: Command line argument: -Dcatalina.base=/opt/tomcat7
Jul 24, 2019 5:09:44 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: Command line argument: -Dcatalina.home=/opt/tomcat7
Jul 24, 2019 5:09:44 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: Command line argument: -Djava.io.tmpdir=/opt/tomcat7/temp
Jul 24, 2019 5:09:44 PM org.apache.catalina.core.AprLifecycleListener lifecycleEvent
INFO: The APR based Apache Tomcat Native library which allows optimal performance in production environments was not found on the java.library.path: /usr/java/packages/lib/aarch64:/usr/lib/aarch64-linux-gnu/jni:/lib/aarch64-linux-gnu:/usr/lib/aarch64-linux-gnu:/usr/lib/jni:/lib:/usr/lib
Jul 24, 2019 5:09:45 PM org.apache.coyote.AbstractProtocol init
INFO: Initializing ProtocolHandler ["http-bio-8080"]
Jul 24, 2019 5:09:45 PM org.apache.coyote.AbstractProtocol init
INFO: Initializing ProtocolHandler ["ajp-bio-8009"]
Jul 24, 2019 5:09:45 PM org.apache.catalina.startup.Catalina load
INFO: Initialization processed in 4199 ms
Jul 24, 2019 5:09:45 PM org.apache.catalina.core.StandardService startInternal
INFO: Starting service Catalina
Jul 24, 2019 5:09:45 PM org.apache.catalina.core.StandardEngine startInternal
INFO: Starting Servlet Engine: Apache Tomcat/7.0.94
Jul 24, 2019 5:09:46 PM org.apache.catalina.startup.HostConfig deployWAR
INFO: Deploying web application archive /opt/tomcat7/webapps/ScadaLTS.war
Jul 24, 2019 5:09:48 PM org.apache.catalina.core.ContainerBase addChildInternal
SEVERE: ContainerBase.addChild: start: 
org.apache.catalina.LifecycleException: Failed to start component [StandardEngine[Catalina].StandardHost[localhost].StandardContext[/ScadaLTS]]
	at org.apache.catalina.util.LifecycleBase.start(LifecycleBase.java:162)
	at org.apache.catalina.core.ContainerBase.addChildInternal(ContainerBase.java:1018)
	at org.apache.catalina.core.ContainerBase.addChild(ContainerBase.java:994)
	at org.apache.catalina.core.StandardHost.addChild(StandardHost.java:652)
	at org.apache.catalina.startup.HostConfig.deployWAR(HostConfig.java:1127)
	at org.apache.catalina.startup.HostConfig$DeployWar.run(HostConfig.java:2020)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.IllegalArgumentException: More than one fragment with the name [spring_web] was found. This is not legal with relative ordering. See section 8.2.2 2c of the Servlet specification for details. Consider using absolute ordering.
	at org.apache.catalina.deploy.WebXml.orderWebFragments(WebXml.java:2338)
[... 58 lines cut ...]
INFO: Deployment of web application directory /opt/tomcat7/webapps/examples has finished in 16,872 ms
Jul 24, 2019 5:10:23 PM org.apache.catalina.startup.HostConfig deployDirectory
INFO: Deploying web application directory /opt/tomcat7/webapps/host-manager
Jul 24, 2019 5:10:33 PM org.apache.catalina.startup.TldConfig execute
INFO: At least one JAR was scanned for TLDs yet contained no TLDs. Enable debug logging for this logger for a complete list of JARs that were scanned but no TLDs were found in them. Skipping unneeded JARs during scanning can improve startup time and JSP compilation time.
Jul 24, 2019 5:10:33 PM org.apache.catalina.startup.HostConfig deployDirectory
INFO: Deployment of web application directory /opt/tomcat7/webapps/host-manager has finished in 10,436 ms
Jul 24, 2019 5:10:33 PM org.apache.catalina.startup.HostConfig deployDirectory
INFO: Deploying web application directory /opt/tomcat7/webapps/manager
Jul 24, 2019 5:10:43 PM org.apache.catalina.startup.TldConfig execute
INFO: At least one JAR was scanned for TLDs yet contained no TLDs. Enable debug logging for this logger for a complete list of JARs that were scanned but no TLDs were found in them. Skipping unneeded JARs during scanning can improve startup time and JSP compilation time.
Jul 24, 2019 5:10:43 PM org.apache.catalina.startup.HostConfig deployDirectory
INFO: Deployment of web application directory /opt/tomcat7/webapps/manager has finished in 9,664 ms
Jul 24, 2019 5:10:43 PM org.apache.catalina.startup.HostConfig deployDirectory
INFO: Deploying web application directory /opt/tomcat7/webapps/ROOT
Jul 24, 2019 5:10:52 PM org.apache.catalina.startup.TldConfig execute
INFO: At least one JAR was scanned for TLDs yet contained no TLDs. Enable debug logging for this logger for a complete list of JARs that were scanned but no TLDs were found in them. Skipping unneeded JARs during scanning can improve startup time and JSP compilation time.
Jul 24, 2019 5:10:52 PM org.apache.catalina.startup.HostConfig deployDirectory
INFO: Deployment of web application directory /opt/tomcat7/webapps/ROOT has finished in 9,421 ms
Jul 24, 2019 5:10:52 PM org.apache.coyote.AbstractProtocol start
INFO: Starting ProtocolHandler ["http-bio-8080"]
Jul 24, 2019 5:10:52 PM org.apache.coyote.AbstractProtocol start
INFO: Starting ProtocolHandler ["ajp-bio-8009"]
Jul 24, 2019 5:10:52 PM org.apache.catalina.startup.Catalina start
INFO: Server startup in 67095 ms

radek2s, 2019-07-29:
It is probably Tomcat error:
 
_INFO: Deploying web application archive /opt/tomcat7/webapps/ScadaLTS.war
Jul 24, 2019 5:09:48 PM org.apache.catalina.core.ContainerBase addChildInternal
**SEVERE: ContainerBase.addChild: start:**
**org.apache.catalina.LifecycleException: Failed to start component** [StandardEngine[Catalina].StandardHost[localhost].StandardContext[/ScadaLTS]]
at org.apache.catalina.util.LifecycleBase.start(LifecycleBase.java:162)_
[...]
_at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.IllegalArgumentException: **More than one fragment with the name [spring_web] was found. This is not legal with relative ordering. See section 8.2.2 2c of the Servlet specification for details. Consider using absolute ordering.**
at org.apache.catalina.deploy.WebXml.orderWebFragments(WebXml.java:2338)
[...]_

Try to delete all files inside webapps dir and then try again pasting ScadaLTS.war file there

jeffersonpimenta, 2019-07-29:
I have try that, fresh instalation of rpi tomcat and java. ScadaBR works just fine in same setup. Any advice? Thanks!

jeffersonpimenta, 2019-08-02:
Update:

I try to put <absolute-ordering/> before </web-app> on /var/lib/tomcat7/webapps/ScadaLTS/WEB-INF/web.xml and got this:

SEVERE: ContainerBase.addChild: start: 
org.apache.catalina.LifecycleException: Failed to start component [StandardEngine[Catalina].StandardHost[localhost].StandardContext[/ScadaLTS]]
	at org.apache.catalina.util.LifecycleBase.start(LifecycleBase.java:162)
	at org.apache.catalina.core.ContainerBase.addChildInternal(ContainerBase.java:1018)
	at org.apache.catalina.core.ContainerBase.addChild(ContainerBase.java:994)
	at org.apache.catalina.core.StandardHost.addChild(StandardHost.java:652)
	at org.apache.catalina.startup.HostConfig.deployWAR(HostConfig.java:1127)
	at org.apache.catalina.startup.HostConfig$DeployWar.run(HostConfig.java:2020)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.IllegalArgumentException: Filter mapping specifies an unknown filter name javamelody
	at org.apache.catalina.core.StandardContext.validateFilterMap(StandardContext.java:3305)
	at org.apache.catalina.core.StandardContext.addFilterMap(StandardContext.java:3270)
	at org.apache.catalina.deploy.WebXml.configureContext(WebXml.java:1357)
	at org.apache.catalina.startup.ContextConfig.webConfig(ContextConfig.java:1300)
	at org.apache.catalina.startup.ContextConfig.configureStart(ContextConfig.java:881)
	at org.apache.catalina.startup.ContextConfig.lifecycleEvent(ContextConfig.java:388)
	at org.apache.catalina.util.LifecycleSupport.fireLifecycleEvent(LifecycleSupport.java:117)
	at org.apache.catalina.util.LifecycleBase.fireLifecycleEvent(LifecycleBase.java:90)
	at org.apache.catalina.core.StandardContext.startInternal(StandardContext.java:5606)
	at org.apache.catalina.util.LifecycleBase.start(LifecycleBase.java:145)
	... 10 more

jeffersonpimenta, 2019-08-07:
It worked using Java7

closed 2019-08-07 by jeffersonpimenta