SCADA-LTS/Scada-LTS #976: the page's footer in some page show in not right place

liaoah, 2019-08-30:
<div class="footer" style="width:100%; text-align:center; ">
    <span>&copy;2012-2018  <fmt:message key="footer.rightsReserved"/></span>
</div>

[image]

liaoah, 2019-08-30:
<div class="footer" style="width:100%; text-align:center; **position: fixed; bottom: 10px;**">

liaoah, 2019-08-30:
<div class="footer" style="width:100%; text-align:center; position: fixed; bottom: 10px;">

liaoah, 2019-08-30:
div class="footer" style="width:100%; text-align:center; **position: fixed; bottom: 10px;"**

closed 2019-08-30 by liaoah

fabiodurao, 2019-09-19:
Page misconfiguration errors usually happen when we add html components to the page, open a tag, and don't close:

eg.:
"[image]"

not putting "</a>" gives this kind of problem, or it can make half the page a link hehe.

If the component is missing, edit the page, and click minimize components, they will appear small (like a lego) and you will be able to find something lost.