SCADA-LTS/Scada-LTS #1000: SCADA LTS - HTTP Status 404

pablorq, 2019-09-25:
After starting a SCADA LTS Docker container as suggested on https://github.com/SCADA-LTS/Scada-LTS with the following command:

`docker run -it -e DOCKER_HOST_IP=`docker-machine ip` -p 81:8080 scadalts/scadalts /root/start.sh
`

...The container works well for some time and then suddenly a "HTTP Status 404" error is shown, like the following:

http://[IP]/ScadaBR/
 
> HTTP Status 404 - /ScadaBR/
> 
> type Status report
> 
> message /ScadaBR/
> 
> description The requested resource is not available.
> Apache Tomcat/7.0.85
> 

Where [IP] is the default Docker IP address and port, most of the times is localhost:81.

Any idea how to solve it?

Thank you in advance!

waqassiddiqi, 2019-09-27:
Do you have anything in logs?

pablorq, 2019-10-07:
**TL;DR**

After some time running the MySQLservice dies. Is necessary to restart it manually with this:
```
docker exec scada service mysql restart
docker exec scada killall tail
```

---
**DETAILED REPORT**

When the error is shown, you can check if all the services are running on the container (in this case named 'scada'):

```
>docker exec scada ps -A
  PID TTY          TIME CMD
    1 ?        00:00:00 start.sh
  790 ?        01:00:22 java
  791 ?        00:01:27 tail
  858 ?        00:00:00 ps
```

As can be seen, no MySQL service is running. This explains why Tomcat is running but SCADA-LTS don't.

You can restart MySQL service inside the container with:

```
docker exec scada service mysql restart
```

After that SCADA-LTS is still down and you have to restart tomcat which can be done in this way:

```
docker exec scada killall tail
```

After a minute or less, all the services are running:
```
>docker exec scada ps -A
  PID TTY          TIME CMD
    1 ?        00:00:00 start.sh
   43 ?        00:00:00 mysqld_safe
  398 ?        00:00:00 mysqld
  481 ?        00:00:31 java
  482 ?        00:00:00 sleep
  618 ?        00:00:00 ps
```

Now SCADA-LTS is running!

closed 2019-10-07 by pablorq

grzesiekb, 2019-10-07:
I think the problem is in the wrong path now it's a ScadaLTS.
http://localhost:8081/ScadaLTS 

## The way I run the docker image
My system is ubuntu 18.04 - docker 18.09.0 

I remove all started machines before.
`docker rm $(docker ps -aq)`

then I delete all the images dockers.
`docker rmi -f $(docker images)`

and I've been running a test version of the docker lately.

`docker run -it -p 8081:8080 scadalts/scadalts:new_develop /root/start.sh`

and in the browser at the address: http://localhost:8081/ScadaLTS/login.htm
I have a working program.

Depending on the local docker configuration and docker version (community/enterprise/linux/windows10/windows7) the url address may be different.

pablorq, 2019-10-07:
No, the path is ok.

In the 'latest' docker image at that time, 'ScadaLTS' was not active but 'ScadaBR'. You can verify this inside the Catalina directory.

In the 'new_develop' image the app was renamed to ScadaLTS inside the Catalina directory an now you can use that path.

**The problem is the mysql service suddenly dying.**

Regards.