SCADA-LTS/Scada-LTS #1010: Problem with build ScadaLTS

mateuszhyski, 2019-10-08:
Two controllers

mateuszhyski, 2019-10-08:
That problem exist when I change branche. I have everything up-to-date - just clean source- just few minutes checkout-ed from github. After this , after build application and change branche to correct branche and build this appears:

org.springframework.beans.factory.BeanCreationException: Error creating bean with name 'org.springframework.web.servlet.mvc.method.annotation.RequestMappingHandlerMapping': Invocation of init method failed; nested exception is java.lang.IllegalStateException: Ambiguous mapping. Cannot map 'viewEditController' method 
protected org.springframework.web.servlet.ModelAndView org.scada_lts.web.mvc.controller.ViewEditController.showForm(javax.servlet.http.HttpServletRequest,java.lang.String) throws java.lang.Exception
to {[/view_edit.shtm],methods=[GET]}: There is already 'viewEditContorller' bean method
protected org.springframework.web.servlet.ModelAndView org.scada_lts.web.mvc.controller.ViewEditContorller.showForm(javax.servlet.http.HttpServletRequest,java.lang.String) throws java.lang.Exception mapped.
        at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.initializeBean(AbstractAutowireCapableBeanFactory.java:1630)
        at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.doCreateBean(AbstractAutowireCapableBeanFactory.java:555)
        at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.createBean(AbstractAutowireCapableBeanFactory.java:483)
        at org.springframework.beans.factory.support.AbstractBeanFactory$1.getObject(AbstractBeanFactory.java:312)
        at org.springframework.beans.factory.support.DefaultSingletonBeanRegistry.getSingleton(DefaultSingletonBeanRegistry.java:230)
        at org.springframework.beans.factory.support.AbstractBeanFactory.doGetBean(AbstractBeanFactory.java:308)
        at org.springframework.beans.factory.support.AbstractBeanFactory.getBean(AbstractBeanFactory.java:197)
        at org.springframework.beans.factory.support.DefaultListableBeanFactory.preInstantiateSingletons(DefaultListableBeanFactory.java:761)
        at org.springframework.context.support.AbstractApplicationContext.finishBeanFactoryInitialization(AbstractApplicationContext.java:867)
        at org.springframework.context.support.AbstractApplicationContext.refresh(AbstractApplicationContext.java:543)
        at org.springframework.web.servlet.FrameworkServlet.configureAndRefreshWebApplicationContext(FrameworkServlet.java:668)
        at org.springframework.web.servlet.FrameworkServlet.createWebApplicationContext(FrameworkServlet.java:634)

, so remove files from HD is not solution of solving this problem. I still find solution.

grzesiekb, 2019-10-08:
I suspect that there are bad pathways or environmental variables on your computer. Tomorrow morning remotely I will can check what is going.

mateuszhyski, 2019-10-08:
Ok. Anyway. Is it possible to put some screnshoots with , environment variables if you have something more than path to tomcat server? Thanks

grzesiekb, 2019-10-08:
To check it, I need access to your computer eg over TeamVier or AnyDesk

mateuszhyski, 2019-10-08:
Ok. One question. Why after build war file we have now ScadaBR.war .In previous action was ScadaLTS.war file. I clean everything and try , maybe alone, solve it.

mateuszhyski, 2019-10-08:
In ScadaBR, we have two controllers ViewEditContorller and ViewEditController. Here is problem.Earlier here was only one controller. I removed one (class file- not needed one) and tried and this works , but I wondering why , here we have two controller classes.

grzesiekb, 2019-10-08:
But it's your fault you didn't get the project into a new dir like I told you.

mateuszhyski, 2019-10-08:
What do you mean about new dir? Not ScadaLTS? Ok, we will try solve it by telco, but thank you.

mateuszhyski, 2019-10-09:
In build file should be additional task which removes all classes (directories with packages) under directory ScadaLTS/WebContent/classes.

grzesiekb, 2019-10-09:
Removing something from ScadaLTS/WebContent/classes 
doesn't make sense, now.

grzesiekb, 2019-10-09:
We work on the branch where the name of the program is ScadaBR.

mateuszhyski, 2019-10-09:
Please see what we have under https://github.com/SCADA-LTS/Scada-LTS/tree/siv/develop/WebContent/WEB-INF/classes under siv/developr branche. Propably ant task clean should clean directories which not exist there(I mean directories with package names). because if we have (for example) two controllers which maps to same url(different file(class) names) like ViewEditContorller and ViewEditController (those two files from two different branches) we will have problem like mine.When we checkout branche, generated class files from previous branche still exist. Generated files not belongs to branche. If you have additional environment setting please put this information like screenshoot. It will be easier.

grzesiekb, 2019-10-09:
You wouldn't have this problem and potentially others like checkout to a new directory and set up the project again in the first order by switching to a new branch from siv/develop. Just like I asked you to.

closed 2019-10-09 by mateuszhyski