SCADA-LTS/Scada-LTS #1035: Not saved points on watch list and name of watch list not saved also on new user.

mateuszhyski, 2019-10-15:
I created new user and selected all points which created by admin for new user( gaved all points to new user).
I moved points to default watchlist(in new user) and changed name of default watchlist and saved this name.
After logout and login (new user) I do not see saved before watchlist and point.[image][image][image]

mateuszhyski, 2019-10-15:
That one bug exist on siv/develop branche

hjbleier, 2020-07-21:
This is just one of many many bugs when you add a user that does not have admin rights to the system. See issues 1114. Also if you try to export your user config it will fail and so will a project export/backup. Every since the "user profile" tab was add to ScadaLTS non-admin users are broken. There are alot of thing that don't work correctly. It is all of the development releases as well as the stable releases.