SCADA-LTS/Scada-LTS #1080: User export has following error!

jantoniaze, 2019-12-13:
User export has following error!
[image]

Limraj, 2019-12-13:
Hey, if you clicked "project export" you started the function of exporting the whole project. If you only need Users, you can select them and click"Generate  JSON". Then copy it to a file. Another thing is that this error you indicated when exporting the project is verifiable.

jantoniaze, 2019-12-16:
The error reported was performing this procedure which reported and presented this error!

hjbleier, 2020-08-27:
@jantoniaze This is a bug that has been in every version since the user profile page was added. As soon as you add a non-admin user it breaks the backup/export, but this is the least of the issues. There are a ton of other problems with sharing watch pages with non-admin users and graphical view pages when there are regular users added to the system. I had to switch back to ScadaBR for my project. Also if a non-admin makes there own watchlist it damn near corrupts all of the other users watchlist, custom reports you name it. If all of your users are all admin stuff will work okay but there is still a ton of bugs with users.

Limraj, 2020-09-03:
Hello, sorry for the late response. I understand that the error scenario is as follows: Select Users and then Generate JSON, on admin user? How many users do you have? 

@hjbleier Non-admin users cannot export the project. This is because such a user should not have access to all data. As for the watchlist problem, the user only sees their created watchlists. Would you suggest any changes here? 
If the problem is something else, we would be grateful if you could describe it to us in detail and we will take care of it. 

We would like to thank everyone for their involvement in the project. :)

Limraj, 2023-06-03:
Hi @jantoniaze and @hjbleier,
The problem should not occur in version 2.6.18 and later. Current version 2.7.5.3.

Regards,
Kamil Jarmusik

closed 2023-06-03 by Limraj