SCADA-LTS/Scada-LTS #1114: Project Export is broken with non-admin users

hjbleier, 2020-02-05:
This affects both the current stable 0.0.9.8 and the most recent development 2.1.0 as well.

Clean install of Scada-LTS 0.0.9.8 with a clean fresh database.
login as admin.
the project export works..
create a new user with admin rights selected.
the project export works..

but the issue is when you create a new user that does not have admin rights on Scada-LTS.
when logged in as admin
the project export fails to generate a zip archive and the page times out.

I have also tried several older versions of Scada-LTS it seems to have been a problem for a long time in all of the recent pre-release and stable releases.

I have tried several versions of tomcat7 to see if that was causing the issues and it does not seem to be the problem. I am running JDK8 with manual installed binary release of tomcat 7.0.81 from apache's archive currently on Ubuntu 18.04 server.

Please let me know if you can replicated this issue as well, I would say it is pretty critical because most people can not have all users with admin rights to the Scada-LTS for security reasons.

hjbleier, 2020-07-02:
Has anyone been able to test this? It is also effecting the ability to share watchlist and graphic displays with a non-admin user. This is a pretty big deal because not all of the users should have admin rights to the system. Please when you have time look into this, let me know if you need more info from me when trying to reproduce the same problem. Thanks

Limraj, 2020-09-03:
Hello, sorry for the late response, non-admin users cannot export the project. This is because such a user should not have access to all data. As for the watchlist problem, the user only sees their created watchlists. Would you suggest any changes here?

hjbleier, 2020-09-13:
@Limraj okay you completely missed what I was saying. I'm going to try to give you the steps to create the exact situation.

1. On a clean Ubuntu desktop/or server pull the latest docker release.
2. Run it.
3. Login as admin/admin 
4. Try to export the system to a backup (it will work like it should)
5. Now create a user (not a admin) and/or user profile
6. Still logged in as admin try the export it is broken now.

Plus a ton of other stuff is broken too, try to create a watchlist as the user it creates a bunch of garbage on the admin accounts watchlist. The graphic pages do not share correctly. The system is only usable if all users are admin and there are still even a few bugs that way too. Just play around and create a few things you will see it is loaded with weird bugs. Almost every thing has user bugs once a non-admin user is created on the system

Limraj, 2020-09-14:
Hey, it all comes down to an exact error description, then we can fix something and get the system back in order. It is a great value for us. :)
We will check the user creation scenario, thanks for the detailed description, and we will check the watchlist. If you could describe in detail any other problems that are hindering your use of the system, we will also take care of them. :)
For the future, you do not need to change the name of the ticket, he will be notified by email about new comments. :)
Thank you for your time and encourage you to actively report bugs. :)

Limraj, 2020-09-14:
Okay, I found the cause of the import problem after creating a non-admin user. I think we will implement a patch in version 2.4.

hjbleier, 2020-09-14:
@Limraj That is awesome thank you! I will try v2.4 when it is released and I will test the other issues I found and provide a detail on how to reproduce them. The others take a bit more steps to make them happen. Thank you for your help

Limraj, 2020-09-14:
We will solve the problems step by step.

Limraj, 2020-09-14:
1. Next problem:
"Plus a ton of other stuff is broken too, try to create a watchlist as the user it creates a bunch of (unnamed watchlist) garbage on the admin accounts watchlist."

1) Admin sees the watch list of all users.
2) To add and rename a watchlist:
[image]

a) click "add new watch list"; (brown)
b) then you need to hover the cursor over "pencil"; (purple)
c) then enter your name in the text box; (green)
d) click "save" (orange)
e) click "Add to watch list" (blue)

If that doesn't fix the problem, let me know. :)

2. "The graphic pages do not share correctly with normal users when they were created as admin even if you set them to readable or setable values for the user."

Scenario:
1) create a grapchical view gv_test;
2) create two profiles: gv_test_read, gv_test_set with permissions read and set for gv_test;
3) create two users:
a) user_gv_test_read (with profile: gv_test_read)
b) user_gv_test_set (with profile: gv_test_set)
4) login  on user_gv_test_read, you can't see gv_test -> error;
5) login on user_gv_test_set, you can't see gv_test -> error;

Does it suit your scenario? I will take a closer look at it.
I will probably open a separate task to fix this error, do you mind?

Limraj, 2020-09-16:
@hjbleier Hey, My English is not the best, so if I have misunderstood something, feel free to write. :)

closed 2023-03-24 by Limraj

Limraj, 2023-04-17:
Hi @hjbleier, 
Most of the things you mentioned have been fixed. 
Latest stable version: [v2.6.18](https://github.com/SCADA-LTS/Scada-LTS/releases/tag/v2.6.18) 
Beta version, many fixes: [v2.7.5.2](https://github.com/SCADA-LTS/Scada-LTS/releases/tag/v2.7.5.2)
If you notice any irregularities, we encourage you to report them.

Regards,
Kamil Jarmusik