SCADA-LTS/Scada-LTS #1135: All dates shall comply with the representation of dates and times defined in ISO8601

Golffies, 2020-02-24:
## Expected Behavior
Dates and times shall look like:
2020-02-24 14:21:17

## Actual Behavior
Dates are shown with slashed instead of dashes:
2020/02/24 14:21:17

## Steps to Reproduce the Problem

1. Open the "Watch list";
2. Select an event in the drop-down menu;
3. Click on a magnifier icon, i.e. "Point details";
4. Look at the "History" table.

## Specifications

  - Version: v2.2.0
  - Platform: Linux
  - Subsystem: open jdk 1.8, mysql 5.7, nodejs 8.12.0

## Correcting issue:
#1128 

## The problem reported by:
@grzesiekb

grzesiekb, 2025-08-30:
Cześć, nie mam uprawnień, aby samodzielnie odpiąć się od tego zadania. Proszę o usunięcie mnie z listy Assignees. Dzięki!
Hi, I don’t have the permissions to unassign myself from this issue. Could you please remove me from the Assignees list? Thanks!

Limraj, 2025-08-31:
Out of date