SCADA-LTS/Scada-LTS #1146: Publisher/HTTP Sender save button generate SQLexception with MariaDB10.4 and Windows10

WagnerDeQueiroz, 2020-03-14:
When creating a new Publisher **HTTP Sender**  on a fresh new ScadaLTS we got a SQLException saving any example:

_The error is:_ 

PreparedStatementCallback; uncategorized SQLException for SQL []; SQL state [null]; error code [0]; Can only convert a byte[] to BINARY, VARBINARY or LONGVARBINARY;
nested exception is 
**java.sql.SQLException**: Can only convert a byte[] to BINARY, VARBINARY or LONGVARBINARY

Found with Tomcat7 / Windows 10 / MariaDB 10.4
Using derby on same instalation, this error not appear...

[image]

grzesiekb, 2020-03-16:
Is the same situation with the docker? 
Docker can also be installed on windows.

WagnerDeQueiroz, 2020-03-17:
I don't have experience with docker. Its possible known what version of each program is used on to install the ScadaLTS on Windows?

grzesiekb, 2020-03-17:
Thank you for your test, if you could write to me on facebook, I will give you contact to the decision maker in matters of work on ScadaLTS.  https://www.facebook.com/ScadaLTS

Brent-Frere, 2020-03-18:
We'll try to reproduce this issue on Thursday.

Brent-Frere, 2020-03-20:
Sorry, I didn't had time for this. I'll try to reproduce this issue during the week-end or on Monday, despite I'm very busy...

Limraj, 2023-06-03:
Hi @WagnerDeQueiroz,
I can't reproduce this bug on 2.7.5.3, it probably doesn't exist anymore.

When it comes to installing the source for windows, we have prepared a complete installer, it also installs the mysql database:
https://github.com/SCADA-LTS/windows-installer

If the problem persists, you can reopen this issue.

Regards,
Kamil Jarmusik

closed 2023-06-03 by Limraj