SCADA-LTS/Scada-LTS #1222: Scroll bar on main page

jeffersonpimenta, 2020-05-13:
Hi,

I have an application to watch solar inverters on ScadaLTS and i'm almost running out of space on Points area:

[image]

In the next versions its possible to implement a scroll bar on it? I know that its possible to adjust the size of it or use zoom out to see hidden points, but it will be nice to have a scroll bar on it!

Thank you very much!

grzesiekb, 2020-05-13:
if you change height You need to refresh window (Ctrl + R) to see the scroll in the tree with list points.

[image]

jeffersonpimenta, 2020-05-13:
Everything works fine with F5! Thanks!

closed 2020-05-13 by grzesiekb