SCADA-LTS/Scada-LTS #1237: Error on database ScadaLTS 2.2.1

jeffersonpimenta, 2020-05-21:
Hi, after trying install ScadaLTS on Rpi3b+  i got the error on catalina.out:

SEVERE: Unable to create initial connections of pool.

After that i tough that i was running on bad mysql configuration, but after open mysql and check if scadalts has created the tables on database i saw that the system create all tables without problems.

[image]

My java version is openjdk version "1.8.0_212" and tomcat7.

Previous i had a mysql error that i manage to solve copying lib files under web-inf/lib to tomcat7/lib. But this time i had to download mysql-connector version 5.1.27, 5.1.38, 5.1.46 and 8.0.20 in order to overcome the error of missing connector

My context.xml is exacly like the example [here](https://github.com/SCADA-LTS/Scada-LTS/wiki/Install-Scada-LTS-on-linux) with my credentials.

My env.properties is like
[image]
[image]

My env.properties.mysql is like
[image]

When i try to access the webpage i got:
[image]

[image]

[image]

IKD if my question was a lame question, but i got success so far last time i try to put this setup on my rpi and got this error on newer version of scadalts.

Thanks everyone!

grzesiekb, 2020-05-22:
In **env.properties** if configure connection in context.xml
```
db.type=mysql
db.datasource=true
db.datasourceName=java:comp/env/jdbc/scadalts

```

And you need to copy the appropriate libraries to the database connections as well as to tomcat.

jeffersonpimenta, 2020-05-22:
I first try using both env.properties and env.properties.mysql. No success. same error.

Then i rename those files to .old and create a new env.properties with only three lines you have mentioned. Same error.

I try to unistall every other java version but 7 and set it on tomcat7. Same error.

[catalina.zip](https://github.com/SCADA-LTS/Scada-LTS/files/4668796/catalina.zip)

TY for your help! :D

grzesiekb, 2020-05-22:
I would need access through Ana Desk, for example. To check it out.

jeffersonpimenta, 2020-05-22:
I could give you access to my computer via teamview. Contact me via jefferson.pimenta@hotmail.com or engjeffersonpimenta@gmail.com

I have both rpi working and not
 
TY!

closed 2020-07-30 by jeffersonpimenta