SCADA-LTS/Scada-LTS #1304: Malfunctioning Enable_all_points button

Limraj, 2020-06-22:
## Expected Behavior
After adding many points and enabling them by the "Enable all points" button, it is possible to change the values ​​for these points. 

## Actual Behavior
After adding many points and enabling them by the "Enable all points" button, you cannot change the value for those points. An error alert appears: "Incorrect format. The point value has not been changed. Error saving point value: dataType=1, dvalue=0.0". 

And mango.log: 
```
(...)
ERROR 2020-06-22 10:26:39,761 (org.scada_lts.mango.service.PointValueService$BatchWriteBehind.execute:578) - Error saving 1 batch inserts. Data lost. 
org.springframework.jdbc.BadSqlGrammarException: PreparedStatementCallback; bad SQL grammar [insert pointValues (dataPointId,dataType,pointValue,ts) values (?,?,?,?)]; nested exception is java.sql.BatchUpdateException: Result consisted of more than one row
(...)
	at org.scada_lts.dao.pointvalues.PointValueDAO.executeBatchUpdateInsert(PointValueDAO.java:527)
	at org.scada_lts.mango.service.PointValueService$BatchWriteBehind.execute(PointValueService.java:554)
	at com.serotonin.mango.rt.maint.BackgroundProcessing$1.run(BackgroundProcessing.java:56)
(...)
WARN  2020-06-22 10:26:43,261 (org.directwebremoting.util.CommonsLoggingOutput.warn:67) - Method execution failed:  
java.lang.RuntimeException: Error saving point value: dataType=1, dvalue=1.0
	at org.scada_lts.mango.service.PointValueService.savePointValueInTrasaction(PointValueService.java:242)
	at org.scada_lts.mango.service.PointValueService.savePointValueImpl(PointValueService.java:153)
(...)
	at org.scada_lts.dao.pointvalues.PointValueDAO.createNoTransaction(PointValueDAO.java:496)
	at org.scada_lts.dao.pointvalues.PointValueDAO.create(PointValueDAO.java:483)
	at org.scada_lts.mango.service.PointValueService.savePointValueImpl(PointValueService.java:280)
	at org.scada_lts.mango.service.PointValueService.savePointValueInTrasaction(PointValueService.java:235)
(...)
```

## Steps to Reproduce the Problem
1) Create Virtual Data Source;
2) Create DataPoint1 and DataPoint2, settable, no change;
3) Click button "Enable all points"
4) On Watchlist change point value;

## Initial Analysis
The "Enable all points" button does not initialize the values ​​in the PointValues ​​table only for one point, but should for all points not activated.

## Specifications

  - Version: v2.2.0
  - Platform: Linux
  - Subsystem: open jdk 1.8, mysql 5.7, nodejs 8.12.0

## The problem reported by:
@Limraj

grzesiekb, 2020-06-23:
In my opinion problem is in trigger with some times not work correctly.
[image]

We can see it below where the inserts normally work when the trigger is removed.

Limraj, 2020-06-23:
Hi, please check if you have a problem with this configuration. In fact, it looks like the topic is more about the time between points launches, if it's too short it breaks down. The problem also occurs on a clean base on dev.

[config_db.txt](https://github.com/SCADA-LTS/Scada-LTS/files/4819333/config_db.txt)

Limraj, 2020-06-25:
The error cannot be repeated. It looks good.

closed 2020-06-25 by Limraj