SCADA-LTS/Scada-LTS #1334: SCADALTS v2.2.1 does not work on linux / SCADALTS v2.2.1 não funciona em linux

germanorocha, 2020-07-09:
SCADALTS v2.2.1 does not work on linux

SCADALTS v2.2.1 não funciona em linux

Limraj, 2020-07-09:
Hi! What exactly is going on?  What is this linux? :)

fabiodurao, 2020-07-09:
It works, I am using exactly this release on a Mint 19.3. The docker is still unable to use, but the manual installation as explained on the github wiki works.
[image]

germanorocha, 2020-07-09:
O Raspberry PI 4 funciona normalmente com o ScadaBR 1.0, porém o ScadaLTS 2.2.1 não opera em nenhuma versão. Eu utilizo o banco de dados MariaDB. Outra solicitação: A versão do ScadaLTS 2.3.0 está sem a compilação WAR. Assim, solicito, por gentileza disponibilizar a versão compilada WAR. Passa a dica da instalação manual, Fábio Durão. O projeto do ScadaLTS está cada vez melhor. Parabéns a todos os colaboradores!

The Raspberry PI 4 works normally with ScadaBR 1.0, however ScadaLTS 2.2.1 does not operate on any version. I use the MariaDB database. Another request: The version of ScadaLTS 2.3.0 is without the WAR build. So, please kindly make available the compiled version WAR. Pass the tip on manual installation, Fábio Durão. The ScadaLTS project is getting better and better. Congratulations to all employees!

Limraj, 2020-07-10:
https://github.com/SCADA-LTS/Scada-LTS/releases/tag/v2.2.1

hjbleier, 2020-07-18:
I currently have Scada-LTS 2.2.1 running on a Rpi3B+, OpenJDK 1.8.0_212, MariaDB 10.3.22, Tomcat 7.0.85. It is working for me following the manual linux installation directions. How are you trying to run it?

fernandoBranqs, 2020-07-26:
> It works, I am using exactly this release on a Mint 19.3. The docker is still unable to use, but the manual installation as explained on the github wiki works.
> [image]

Fabio, are you talking about the explanation on this link ?
https://github.com/SCADA-LTS/Scada-LTS/wiki/Install-Scada-LTS-on-linux

Limraj, 2023-06-03:
Hi @germanorocha,
During this time, we improved the operation of Scada on docker, we use the docker-compose configuration:
https://github.com/SCADA-LTS/Scada-LTS/blob/develop/docker-compose.yml

All tags:
https://hub.docker.com/r/scadalts/scadalts/tags

It is also worth trying the latest version 2.6.18 (docker tag v2.6.18 or https://github.com/SCADA-LTS/Scada-LTS/releases/tag/v2.6.18), or beta 2.7.5.3 (tag v2.7.5.3 or https://github.com/SCADA-LTS/Scada-LTS/releases/tag/v2.7.5.3).

Regards,
Kamil Jarmusik

closed 2023-06-03 by Limraj