SCADA-LTS/Scada-LTS #1339: ScadaBR vs ScadaLTS

fernandoBranqs, 2020-07-26:
I am confused.

From what I'm seeing, ScadaLTS derived from ScadaBR, but this site talks about the two versions how existing two versions in parallel.

I say this because the documentation several times talks about downloading ScadaBR.war and also ScadaLTS.war.

On the other hand, the changelog only introduces new versions of ScadaBR.war

Are both the same version ?

I'd appreciate for a help.

fernandoBranqs, 2020-07-27:
I saw that ScadaLTS.war footprint (1.55 Mb) is very small compared with ScadaBR.war (67 Mb)

How should i use them ?

Thank you very much.

grzesiekb, 2020-09-08:
Download ScadaBR.war file from last release https://github.com/SCADA-LTS/Scada-LTS/releases/tag/v2.3.1
For the time being, we have kept the name ScadaBR.war as a result of the deployment of which we have where the links with the name are hard-coded and for this we do not change the name.

closed 2020-09-08 by grzesiekb