SCADA-LTS/Scada-LTS #1360: Replace waffle.io with issue.sh

fabiodurao, 2020-08-24:
Since last year when waffle.io ended, there has been no update to a graphical project management tool. I researched and discovered the issue.sh (https://issue.sh) which is free for an open source project, I suggest its implementation to help colleagues who are not from the development area to follow the evolution of the project.

radek2s, 2021-01-15:
Since Scada-LTS 2.5.0 version we have started using [GitHub Projects](https://github.com/SCADA-LTS/Scada-LTS/projects/19) to provide a Kanban board to our project. Now backlog for iteration is descripted there and it is much easier to be up-to-date with progress of development. Advantage of that solution is integration with our existing accounts so developers do not have to create another profiles on the 3-rd party sites. Now people that are not developer can see status of each task that is going to be provided.

closed 2021-01-15 by radek2s

Golffies, 2021-01-16:
Hello Radek,

> Since Scada-LTS 2.5.0 version we have started using GitHub Projects to provide a Kanban board to our project. Now backlog for iteration is descripted there and it is much easier to be up-to-date [...]

I love that. It delivers a higher level view than browsing into tasks and commits, and is ideal to communicate what we are doing in near real-time to our client, should we need to do so. One and a half year ago, Grzeszek and I tried to achieve something similar, based on ZenHub, interfaced with GitHub. Perhaps it failed because of a lack of our motivation to populate the Kanban board.

That time I am going to have a deeper look at it.

Is the Kanban board populated with tasks automatically from actions performed on our main repository (I am afraid not), or should each one create tasks manually into the board to make them appear there (I guess so) ?  

Thanks!

Golffies, 2021-01-16:
Hello Radku!

> Since Scada-LTS 2.5.0 version we have started using GitHub Projects to provide a Kanban board to our project. Now backlog for iteration is descripted there and it is much easier to be up-to-date [...]

I love that. It delivers a higher level view than browsing into tasks and commits, and is ideal to communicate what we are doing in near real-time to our client, should we need to do so. We may then have to enhance the description given to the tasks, we will see. One and a half year ago, Grzeszek and I tried to achieve something similar, based on ZenHub, interfaced with GitHub. Perhaps it failed because of a lack of our motivation to populate the Kanban board.

That time I am going to have a deeper look at it.

Is the Kanban board populated with tasks automatically from actions performed on our main repository (I am afraid not), or should people create tasks manually into the board to make them appear there (I guess so) ?  

Thanks!