SCADA-LTS/Scada-LTS #1384: Multiple XSS Vulnerabilities Within ScadaLTS Web Interface

Galapag0s, 2020-09-15:
While reviewing the ScadaBR Web Interface, I identified multiple XSS vulnerabilities. These vulnerabilities range in the amount of user interaction that is required. 

The first XSS vulnerability lies within the SQL query panel. The output of the SQL queries within this panel is not properly sanitized. In worst case scenarios, this could allow arbitrary JavaScript to execute under the permissions of an administrator. I was able to successfully fire an alert box by performing the below steps.

1. Log in as a low level user.
2. Navigate to the "Users" panel.
3. Append "<script>alert(1);</script>" to the end of the user's email.
4. Log out and log back in with an "admin" account.
5. Navigate to the "SQL" panel.
6. Execute the command "SELECT * FROM users;"
7. A JavaScript alert box will fire.

The output returned from these SQL queries should be properly sanitized.

The second XSS vulnerability required less user interaction. This attack could be performed by a low level account. If a low level user has access to any data point they are able to add arbitrary "User Notes." Unfortunately, these user notes are not properly sanitized. As such, a low level user can enter arbitrary JavaScript which will execute when a page the affected data point page is loaded. Within this application, arbitrary JavaScript can have serious impact on the systems confidentiality, integrity, and availability. Below are steps to reproduce the attack.

1. Create an active data source as an admin and provide "read" rights to a low level user. Log out of the admin account.
2. Log in as a low level user.
3. From the watch list click "Point Details."
4. Click "Add a Comment" under "User Notes."
5. Enter "<script>alert(1);</script>" into the text field.
6. Log out of the low level user account.
7. Log in to the system as an administrator and click "Point Details."
8. An alert box will fire.

The third XSS vulnerability requires a considerable amount of interaction from a user, but is still vulnerable. For this XSS vulnerability to work, the ScadaLTS server must be polling a remote web server for an image. If a threat actor was able to replace the contents of the remote web server's image with an HTML file, arbitrary JavaScript could execute. In order to execute, the user would need to load the image directly and remove the "?w=#&h=#" parameters. If this is done, arbitrary HTML and JavaScript can be executed. The risk related to this vulnerability is admittedly low, but present.

Limraj, 2020-09-15:
Hi, thanks for accurately presenting such a serious error. we will probably implement:
https://owasp.org/www-project-java-encoder/

AdisonCavani, 2022-02-25:
Fixed in #2103

closed 2022-02-25 by AdisonCavani