SCADA-LTS/Scada-LTS #1534: release package  war  have  redundancy and  conflict jar

huanghe-1, 2021-02-09:
example, https://github.com/SCADA-LTS/Scada-LTS/releases/download/v2.5.1/ScadaBR.war
mysql-connector-java have version 3.1.8, 5,1,7, 5,1,38 at same time.
spring have version 3.2.17.RELEASE, 4.3.16.RELEASE  at same time

arturro96, 2021-02-09:
We have two versions of Spring because of seroUtils.jar which uses methods from spring 3.2.17 which spring 4.3.16 doesn't have. In near future we're planning to separate seroUtils with its dependencies, so there won't be conflict jars.

closed 2021-04-19 by SoftQ

huanghe-1, 2021-04-20:
for  think long-term，do you have plan to replace the  seroUtils.jar?

arturro96, 2021-04-20:
Yes, that's our plan, to remove seroUtils.jar from Scada. We will be doing it step by step and hopefully we will be able to get rid of this library.

commit 37008ff6 (2021-04-19): build.gradle fix #1534