SCADA-LTS/Scada-LTS #1582: Problems when trying to run the docker

ejs94, 2021-03-17:
After following the tutorial to run an application on the docker, the following log appears to me:

standard_init_linux.go:219: exec user process caused: exec format error

My inspect is:

{
    "AppArmorProfile": "",
    "Args": [],
    "Config": {
        "AttachStderr": false,
        "AttachStdin": false,
        "AttachStdout": false,
        "Cmd": [
            "/root/start.sh"
        ],
        "Domainname": "",
        "Entrypoint": null,
        "Env": [
            "PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin"
        ],
        "ExposedPorts": {
            "8080/tcp": {}
        },
        "Hostname": "48e595fa5339",
        "Image": "scadalts/scadalts:latest",
        "Labels": {},
        "OnBuild": null,
        "OpenStdin": false,
        "StdinOnce": false,
        "Tty": false,
        "User": "",
        "Volumes": null,
        "WorkingDir": ""
    },
    "Created": "2021-03-17T16:42:50.32907133Z",
    "Driver": "overlay2",
    "ExecIDs": null,
    "GraphDriver": {
        "Data": {
            "LowerDir": "/var/lib/docker/overlay2/96863625f4e97e64c4ea2a12258123f8517754b6b9ff9a7eeb1dc1740322aa86-init/diff:/var/lib/docker/overlay2/15c1d724c7f1aa3dec93e0bad207daf7f63c82b6142f0dea396b1f13f3f0e90f/diff:/var/lib/docker/overlay2/8c21a2bb10258dd11cc00ff7be920db8ecc3c271b916ede38aad87d2338973ee/diff:/var/lib/docker/overlay2/13bbaa9c8e5b85ce7fa267de0377136d7ceffcf6aad6f90779b51e8f15e894c7/diff:/var/lib/docker/overlay2/a8eb27eb109fdfecff26490199cbd8de7399057aa9fbf5a1e6e79ef4b1c09978/diff:/var/lib/docker/overlay2/7a776a14472336a14638b914d681989c855cf14e8582e20d11b6af30ca4292ad/diff:/var/lib/docker/overlay2/ec1db8ab1e3a69dcfa954d387731df1e448c3ef1532c83bb026803d257a2f04b/diff:/var/lib/docker/overlay2/1390266fac8298dbbf1a399e8bcd8807a7da10e23b1e7d4e5d03642b2700b9cb/diff:/var/lib/docker/overlay2/026fa2e4aea191b7bd4be0c15d0e18ced1809095c03e6691d364367a948deee4/diff:/var/lib/docker/overlay2/c9c7bc6a2607332b77fe1fce82a08c900da70dca920027b26a8acfb262995423/diff:/var/lib/docker/overlay2/78e9b57cef9dd792d6bc2eab039f812a2f25b676b4d4730a2810e4f1375ae71a/diff:/var/lib/docker/overlay2/9bd09e350b2d3581b6b9ada6f1a3622b7dec549f9bc93b41259aa5ddacd43b43/diff:/var/lib/docker/overlay2/7f566e46bdf0f08abbbcde94bd09b039d7484e48187e982b4c2c225235936986/diff:/var/lib/docker/overlay2/fd9e8d123b3c407e420ed9a60edcbd3de58dd6a0cf7df102465288ac815d9533/diff:/var/lib/docker/overlay2/5edc64742c67303226e9c54e9c4b36fa1d8ebb6c9bab2fc22c9b75f5a180d8fd/diff",
            "MergedDir": "/var/lib/docker/overlay2/96863625f4e97e64c4ea2a12258123f8517754b6b9ff9a7eeb1dc1740322aa86/merged",
            "UpperDir": "/var/lib/docker/overlay2/96863625f4e97e64c4ea2a12258123f8517754b6b9ff9a7eeb1dc1740322aa86/diff",
            "WorkDir": "/var/lib/docker/overlay2/96863625f4e97e64c4ea2a12258123f8517754b6b9ff9a7eeb1dc1740322aa86/work"
        },
        "Name": "overlay2"
    },
    "HostConfig": {
        "AutoRemove": false,
        "Binds": [],
        "BlkioDeviceReadBps": null,
        "BlkioDeviceReadIOps": null,
        "BlkioDeviceWriteBps": null,
        "BlkioDeviceWriteIOps": null,
        "BlkioWeight": 0,
        "BlkioWeightDevice": null,
        "CapAdd": [
            "AUDIT_WRITE",
            "CHOWN",
            "DAC_OVERRIDE",
            "FOWNER",
            "FSETID",
            "KILL",
            "MKNOD",
            "NET_BIND_SERVICE",
            "NET_RAW",
            "SETFCAP",
            "SETGID",
            "SETPCAP",
            "SETUID",
[... 168 lines cut ...]
            "Type": 1,
            "UserAccesses": [],
            "TeamAccesses": [],
            "Public": false,
            "AdministratorsOnly": true,
            "System": false
        }
    },
    "ProcessLabel": "",
    "ResolvConfPath": "/var/lib/docker/containers/48e595fa533917e185658c5c50c949b3cc251776d0f04e23b01b467075dcd54d/resolv.conf",
    "RestartCount": 18,
    "State": {
        "Dead": false,
        "Error": "",
        "ExitCode": 1,
        "FinishedAt": "2021-03-17T16:51:56.922605024Z",
        "OOMKilled": false,
        "Paused": false,
        "Pid": 0,
        "Restarting": true,
        "Running": true,
        "StartedAt": "2021-03-17T16:51:56.834884605Z",
        "Status": "restarting"
    }
}

Limraj, 2021-03-17:
Hello. Are you trying to do this in the console or in a script?

Maybe try it like this:
`docker run -it -p 81:8080 scadalts/scadalts /root/start.sh`

Maybe it will help you? [exec-user-process-caused]( https://stackoverflow.com/questions/42494853/standard-init-linux-go178-exec-user-process-caused-exec-format-error)

ejs94, 2021-03-18:
Hi @Limraj, maybe i found the problem. I didn't mention that I'm trying to open it from a raspberry pi docker.
So after trying:
`docker run -it -p 81:8080 scadalts/scadalts /root/start.sh`

This error appears:
>  WARNING: The requested image's platform (linux/amd64) does not match the detecte            d host platform (linux/arm/v7) and no specific platform was requested

So i try:
`docker run -it -p 81:8080 --platform linux/arm/v7  scadalts/scadalts /root/start.sh`

This error appears:
> docker: Error response from daemon: image with reference scadalts/scadalts was found but does not match the specified platform: wanted linux/arm/v7, actual: linux/amd64.
See 'docker run --help'.

I need to mount an image based on a kernel arm, right?

Limraj, 2021-03-19:
Hey that's right, we currently support linux/amd64. We will consider extending to other architectures, including ARM v7.

closed 2021-03-22 by ejs94