SCADA-LTS/Scada-LTS #1590: Exception système!

gueyefat, 2021-03-25:
Bonjour,
Je chargais des données depuis mon ScadaBR vers ScadaLTS via import/export par script. En vérifiant si mes données s'afficher dans mes sources de données je tombe sur une page d'exception que voilà:
**Exception système!

Le serveur a rencontré une exception lors du traitement de votre dernière requête. L'exception a été enregistrée et les administrateurs système seront informés de ce problème. Vous pouvez continuer à utiliser le site. Nous nous excusons pour le dérangement.

Veuillez fournir le code suivant à l'administrateur pour identifier le problème:
20210325083616576
© 2012-2021 Scada-LTS Tous droits réservés.**

Pourriez vous m'aider à trouver pourquoi cela s'affiche t-il?

Merci d'avance!

Golffies, 2021-03-25:
Bonjour,
on peut envisager deux pistes:

* dans l’hypothèse où votre SCADA-BR utiliserait une base de données mySQL, on peut tenter:
  * d'en faire une copie
  * d'installer SCADA-LTS en lui faisant utiliser cette copie, plutôt que de démarrer sur une base de données vierge
  * SCADA-LTS va alors tenter de mettre à jour la structure de la base de donnée SCADA-BR vers SCADA-LTS
  * enfin, de faire une copie de votre répertoire catalogue, avec les fonds d'écran

* l'autre option serait:
  * de faire une installation nouvelle de SCADA-LTS avec une base de donnée mySQL vierge
  * de copier tout pareillement votre répertoire catalogue, avec les fonds d'écran
  * d'exporter sous forme de fichiers .json vos points de mesure, vos vues, éventuellement vos autres éléments déjà définis dans SCADA-BR
  * d'importer ces fichiers .json un par un, et non sous la forme d'un "projet" complet

Vous connaissez l'adage, le diable se cache dans les détails. En fonction de votre configuration, on s'y prendra au cas par cas de manière différente. A vous de juger si ça justifie qu'on y passe ensemble quelques heures, sous la forme d'un support technique commercial.

En espérant que votre import se passera bien !

gueyefat, 2021-03-25:
Merci de votre réponse!
C'est la seconde option que j'ai utilisé et ça m'affichait cette exception. Alors j'ai supprimé la base données et j'ai redémarrer ScadaLTS. Puis j'ai commencé par importer les datasources. Jusque là aucun soucis, j'ai poursuivie par les datapoints mais vue leur nombre ; je les importais par 6000. Arrivée à un instant , je vérifie ma page de Data sources pour voir si les données était bien chargées c'est alors qu'apparait cette exception.
Est ce que la première option s'agit d'exporter le projet ScadaBR? Si c'est cela, j'ai essayé de l'exporter mais la page qui s'ouvrait lorsque j'appuyais sur le bouton **Copier** tourner en vain
[image]

Golffies, 2021-03-25:
Non, la première option ne consiste pas à exporter puis importer le projet. Elle consiste plus brutalement à faire démarrer SCADA-LTS sur une base de données structurée par SCADA-BR, et à laisser SCADA-LTS retrouver ses petits et mettre la base à jour. A tester bien sûr sur une copie de la base.

La fonction export-import de projet est connue pour ne pas marcher. Nous venons de corriger la fonction d'export sur SCADA-LTS (#993); ce sera disponible dans la version 2.6. Malheureusement, ça ne vous sera pas directement utile ici. En production, l'export nous était nécessaire pour un travail de sécurité annexe, permettant de tenir un relevé depuis l'extérieur, des utilisateurs introduisant des modifications à la configuration de SCADA-LTS.

Malheureusement, nous sommes contraints d'être économes de notre temps, et de l'utiliser là où on nous le demande. Nous avons conservé trace que le problème d'import de projet (#1555), lui, devra encore être résolu dans le futur.

Avec quelques dizaine de milliers de points, c'est une belle bête que vous avez là. Ça représente combien de sites distants, pour cumuler un tel volume ?

gueyefat, 2021-03-25:
D'accord si je comprend bien, j'exporte ma base de donnée ScadaBR depuis Mysql Workbench que j'utilise, je l'importe en le renommant en scadalts et puis l'installe ScadaLTS. Est ce bien cela?
On travaille avec 6000 datasources presque!

gueyefat, 2021-03-26:
Bonjour!

J'ai testé l'option 1 qui était de faire démarrer ScadaLTS avec la copie de la base de données de ScadaBR. L'exception n'a pas été levée. Auriez-vous une solution pour m'aider à migrer vers scadaLTS avec ma base de données actuelle?

Merci d'avance!

Golffies, 2021-03-26:
Bonjour,

Nous nous sommes bien compris. A ceci près que nous conseillerions de faire l'exercice avec mysqldump, plutôt qu'avec MySQL Workbench, qui nous semble introduire parfois des conversions de caractères spéciaux mal maitrisées. Pour être bien explicite, voici ce que pourraient être les étapes:

- à l'aide de mysqldump, extraire un dump de votre base de données SCADA-BR; il sera au format texte;
`mysqldump -u root -proot namedb > scada-br_dump.sql`

- créer une nouvelle base de données vierge, qui sera utilisée par SCADA-LTS; son nom reste à votre convenance;

- importer le dump dans cette base de données vierge;
`mysql -u root -proot newdb_scada-lts < scada-br_dump.sql`

- configurer SCADA-LTS pour qu'il utilise cette base de données.

De votre message aujourd'hui, je comprends que c'est le chemin que vous avez plus ou moins suivi. Donc, si ça ne marche pas, il faut débuguer pour voir pourquoi. Je ne peux guère vous proposer d'aller plus loin sur mon temps personnel.

gueyefat, 2021-04-09:
Merci de votre aide Golffies!
C'était un problème de versionning! La version du Mysql server (5.6) d'où je copiais la base de donnée n'etait pas prise en charge par les connecteurs(8.0.11 et 8.0.23) présents dans le serveur(tomcat) utilisés par ScadaLTS. J'ai simplement remplacé ces derniers par ceux présents dans le lib da mon application ScadaLTS.

closed 2021-04-09 by gueyefat