SCADA-LTS/Scada-LTS #1684: readme document docker startup url wrong

huanghe-1, 2021-05-06:
the lastest docker image, startup url is  ScadaBR not ScadaLTS，change  the readme document or the docker.

Type in docker IP adrress in the browser with port 81 and /ScadaBR / sufix. Example: 
http://10.0.75.2:81/ScadaBR /

and,how to pull a requests,tip  no permission

radek2s, 2021-05-06:
Hello @huanghe-1! 
Check our [Wiki page](https://github.com/SCADA-LTS/Scada-LTS/wiki/Run-ScadaLTS-on-docker---instruction) with tutorial how to use docker with our Scada-LTS

huanghe-1, 2021-05-06:
Wiki page is right, readme  mistake

radek2s, 2021-05-06:
Readme will be soon refactored, but we try migrate to Wiki page that is much easier to maintain. Readme file will be redirecting to specific sections on our Wiki and there you will be able to find the latest updates.

closed 2021-05-06 by radek2s