SCADA-LTS/Scada-LTS #1687: Docker image (latest) is not functional

fabiodurao, 2021-05-06:
**Describe the bug**
The installation of the latest release available on the docker is not working, it had tested version 2.6 and had worked normally. Tomcat works but ScadaBR / ScadaLTS does not work. Apparently it is a permission issue in /var/run/mysqld .

**To Reproduce**
Steps to reproduce the behavior:
1. go to the terminal and type as explained in the wiki:
2. sudo docker pull scadalts/scadalts:latest
3. sudo docker run -p 81:8080 scadalts/scadalts:latest
4. go to browser and type http://localhost:81/ScadaBR/

**Expected behavior**
The login window was expected to appear, but it only displays the 404 error.

**Logs in terminal**
---------------------------------------------------------------------
fabio@pop-os:~$ sudo docker run -p 81:8080 scadalts/scadalts:latest
chown: cannot access '/var/run/mysqld': No such file or directory
 * Starting MySQL database server mysqld
   ...done.
mysql: [Warning] Using a password on the command line interface can be insecure.
Tomcat started.
Tomcat started.
tail: cannot open '100' for reading: No such file or directory
==> /opt/scadalts/logs/catalina.out <==

The last packet successfully received from the server was 1 milliseconds ago.  The last packet sent successfully to the server was 0 milliseconds ago.]
May 06, 2021 4:28:03 PM org.apache.catalina.startup.HostConfig deployDirectory
INFO: Deployment of web application directory /opt/scadalts/webapps/host-manager has finished in 56 ms
May 06, 2021 4:28:03 PM org.apache.coyote.AbstractProtocol start
INFO: Starting ProtocolHandler ["http-bio-8080"]
May 06, 2021 4:28:03 PM org.apache.coyote.AbstractProtocol start
INFO: Starting ProtocolHandler ["ajp-bio-8009"]
May 06, 2021 4:28:03 PM org.apache.catalina.startup.Catalina start
INFO: Server startup in 5105 ms
May 06, 2021 4:28:58 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: Server version:        Apache Tomcat/7.0.85
May 06, 2021 4:28:58 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: Server built:          Feb 7 2018 18:52:33 UTC
May 06, 2021 4:28:58 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: Server number:         7.0.85.0
May 06, 2021 4:28:58 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: OS Name:               Linux
May 06, 2021 4:28:58 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: OS Version:            5.11.0-7614-generic
May 06, 2021 4:28:58 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: Architecture:          amd64
May 06, 2021 4:28:58 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: Java Home:             /usr/lib/jvm/java-8-openjdk-amd64/jre
May 06, 2021 4:28:58 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: JVM Version:           1.8.0_292-8u292-b10-0ubuntu1~16.04.1-b10
May 06, 2021 4:28:58 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: JVM Vendor:            Private Build
May 06, 2021 4:28:58 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: CATALINA_BASE:         /opt/scadalts
May 06, 2021 4:28:58 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: CATALINA_HOME:         /opt/scadalts
May 06, 2021 4:28:58 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: Command line argument: -Djava.util.logging.config.file=/opt/scadalts/conf/logging.properties
May 06, 2021 4:28:58 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: Command line argument: -Djava.util.logging.manager=org.apache.juli.ClassLoaderLogManager
May 06, 2021 4:28:58 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: Command line argument: -Djdk.tls.ephemeralDHKeySize=2048
May 06, 2021 4:28:58 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: Command line argument: -Dignore.endorsed.dirs=
May 06, 2021 4:28:58 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: Command line argument: -Dcatalina.base=/opt/scadalts
May 06, 2021 4:28:58 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: Command line argument: -Dcatalina.home=/opt/scadalts
May 06, 2021 4:28:58 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: Command line argument: -Djava.io.tmpdir=/opt/scadalts/temp
[... 884 lines cut ...]
	at com.mysql.jdbc.ExportControlled.transformSocketToSSLSocket(ExportControlled.java:149)
	... 48 more

May 06, 2021 4:29:01 PM org.apache.catalina.core.NamingContextListener addResource
WARNING: Failed to register in JMX: javax.naming.NamingException: Unexpected exception resolving reference [Root exception is com.mysql.jdbc.exceptions.jdbc4.CommunicationsException: Communications link failure

The last packet successfully received from the server was 0 milliseconds ago.  The last packet sent successfully to the server was 0 milliseconds ago.]
May 06, 2021 4:29:01 PM org.apache.catalina.startup.HostConfig deployDirectory
INFO: Deployment of web application directory /opt/scadalts/webapps/host-manager has finished in 57 ms
May 06, 2021 4:29:01 PM org.apache.coyote.AbstractProtocol start
INFO: Starting ProtocolHandler ["http-bio-8080"]
May 06, 2021 4:29:01 PM org.apache.coyote.AbstractProtocol start
INFO: Starting ProtocolHandler ["ajp-bio-8009"]
May 06, 2021 4:29:01 PM org.apache.catalina.startup.Catalina start
INFO: Server startup in 3660 ms

---------------------------------------------------------------------

**Desktop:**
 - OS: Pop OS 20.04
 - Browser: chrome
 - Version: 90

**Additional context**
Add any other context about the problem here.

ourique98, 2021-05-06:
I have the same problem. 
when i instaled docker pull scadalts/scadatls, and install container with: 
docker run -d -p 81:8080 scadalts/scadalts /root/start.sh

in the browser occurred Error 404

SoftQ, 2021-05-06:
A scadalts/scadalts is for an old and stable version. We recommend using
newer one which has several fixes and features. Hopefully tomorrow we will
release a 2.6.3 version - you will get it with "docker pull
scadalts/scadalts:release" command. This release will be  available later
as "scadalts/scadalts".

Best regards !

ourique98, 2021-05-06:
ok today i tried with scadalts/scadalts:release and the same problem occurred.

i wait until tomorrow for the new post to the "release"

thanks alot

Ourique

Limraj, 2021-05-06:
https://ubuntu.com/security/notices/USN-4892-1
This is the likely cause of the problem, docker images created after April 27, 2021 have stopped working.

Currently, the docker image is created with the version:
`JVM Version: 1.8.0_292-8u292-b10-0ubuntu1~16.04.1-b10`
Before:
`JVM Version: 1.8.0_282-8u282-b08-0ubuntu1~16.04-b08`

The system version for created images before and after this date is identical:
`VERSION="16.04.7 LTS (Xenial Xerus)"`
and mysql server version:
`Server version: 5.7.33-0ubuntu0.16.04.1 (Ubuntu)`

Images created before 27 are running, the last one running good is 1607:
https://hub.docker.com/r/scadalts/scadalts/tags?page=1&ordering=last_updated

Limraj, 2021-05-08:
Now it should work for release. Attention, For local connection of the application to the database, the useSSL option has been disabled(jdbc connection properties useSSL=false). We'll decide next week how we approach this, but at the moment it should be okay.

fabiodurao, 2021-05-10:
Thanks Limraj, it worked using

sudo docker pull scadalts/scadalts:release
sudo docker run -d -p 81: 8080 scadalts/scadalts:release

It is not the last release, so what shows in the ScadaLTS settings is 2.6.0 but I can test it until they can solve the java issue.

closed 2021-06-16 by Limraj