SCADA-LTS/Scada-LTS #1693: Cannot read property 'style' of null - Virtual Data Source

fabiodurao, 2021-05-10:
**Describe the bug**
The browser is unable to load the style property in the virtual data source. This error is a bug inherited since the ScadaBR 1.0 version developed in Brazil, ScadaLTS is a branch of ScadaBR 1.1 that also had this error and has not yet been fixed.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to ScadaBR / ScadaLTS.
2. In the Datasource section, choose the Virtual DataSource option and click create.
3. Give the datasource a name and update period and save.
4. When clicking to create a new datapoint this error appears. The same happens if you have changed pages and when you try to edit a virtual datapoint when you return.
Note: In some browsers and PC's this error may take a few seconds to appear.

**Expected behavior**
The creation and editing of the datapoint was expected to happen normally as it does in other Datasources.

**Screenshots**
[image]

**Desktop:**
- OS: Pop OS 20.04
- Browser: chrome
- Version: 90

**Additional context**
This error is apparently simple, it seems that just ok in the popup would solve, but it is a little more complex, if you try to create a numeric datapoint of the attractor type, the "Attraction point" field does not load other datapoints. To get around this in this case, you need to create a numeric data point with any parameter, just to agree to save the datapoint. Then you need to enter the page again, give an ok in the error popup, edit some datapoint of the virtual datasource, it will not show the "change type" field, but just select it manually and then you will be able to find the "Attraction point".

The problem seems to be related to the "change type" field.

celsou, 2021-06-14:
The fix for this bug is to edit the `WEB-INF/jsp/dataSourceEdit/editVirtual.jsp` file and modify the first lines of the **changeChangeType()** function, which should look like this: 

~~~javascript
function changeChangeType() {
      if (currentChangeType) {
          // Close the current change type div.  
          hide(currentChangeType);
      }

      if ($get("changeTypeId").length) {
          var changeTypeId = "divCH" + $get("changeTypeId");
          currentChangeType = changeTypeId;
          // Open the selected type.
          show(changeTypeId);
      }
     
      // Update the values.
      var locator = currentPoint.pointLocator;
      $set("alternateBooleanChange.startValue", locator.alternateBooleanChange.startValue);
      $set("brownianChange.min", locator.brownianChange.min);
      $set("brownianChange.max", locator.brownianChange.max);
      $set("brownianChange.maxChange", locator.brownianChange.maxChange);
      $set("brownianChange.startValue", locator.brownianChange.startValue);
      $set("incrementAnalogChange.min", locator.incrementAnalogChange.min);
      $set("incrementAnalogChange.max", locator.incrementAnalogChange.max);
      $set("incrementAnalogChange.change", locator.incrementAnalogChange.change);
      $set("incrementAnalogChange.roll", locator.incrementAnalogChange.roll == "true");
      $set("incrementAnalogChange.startValue", locator.incrementAnalogChange.startValue);
      refreshValueList("incrementMultistate", locator.incrementMultistateChange.values);
      $set("incrementMultistateChange.roll", locator.incrementMultistateChange.roll == "true");
      $set("incrementMultistateChange.startValue", locator.incrementMultistateChange.startValue);
      $set("noChange.startValue", locator.noChange.startValue);
      $set("randomAnalogChange.min", locator.randomAnalogChange.min);
      $set("randomAnalogChange.max", locator.randomAnalogChange.max);
      $set("randomAnalogChange.startValue", locator.randomAnalogChange.startValue);
      $set("randomBooleanChange.startValue", locator.randomBooleanChange.startValue);
      refreshValueList("randomMultistate", locator.randomMultistateChange.values);
      $set("randomMultistateChange.startValue", locator.randomMultistateChange.startValue);
      $set("analogAttractorChange.maxChange", locator.analogAttractorChange.maxChange);
      $set("analogAttractorChange.volatility", locator.analogAttractorChange.volatility);
      $set("analogAttractorChange.attractionPointId", locator.analogAttractorChange.attractionPointId);
      $set("analogAttractorChange.startValue", locator.analogAttractorChange.startValue);
}
~~~

Limraj, 2023-03-26:
fixed in #2467

closed 2023-03-26 by Limraj