SCADA-LTS/Scada-LTS #1694: Modbus Serial is not working

fabiodurao, 2021-05-10:
**Describe the bug**
The Modbus Serial protocol/datasource is not working, it does not load the COM port or ttyS0.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the Datasource section and choose the Datasource Serial option and click create.
2. You will see that it does not load the serial port, it shows an error from the rxtxserial library.

**Expected behavior**
On windows it was expected to show all existing or compatible COM (on Windows) or ttySX (on Linux) ports.

**Screenshots**
[image]

**Desktop:**
- OS: Pop OS 20.04
- Browser: chrome
- Version: 90

**Additional context**
This is a bug inherited from ScadaBR 1.1 developed in Brazil, one of the things that prevent many users from migrating to ScadaLTS is the lack of this protocol, it is only available in ScadaBR 1.0 developed in Brazil, which is a very outdated version with java 6 and tomcat 6 yet.

The version of ScadaBR 1.1 was developed at the request of a Brazilian health company and this occurred at the same time that the Canadian project had its license closed, as we did not have the source code for the serial modbus, it was disabled or removed, I think it was the first option because some users were able to reactivate it manually in ScadaBR 1.1. This was discussed in the ScadaBR forum (http://forum.scadabr.com.br/t/erro-no-driver-modbus-serial-na-versao-do-scadabr-1-1-0-rc-do-branches/1307/4).

Although personally I do not use the serial modbus protocol [and when I need to use it, I prefer to use a serial modbus to modbus tcp converter (gateway) as I find the serial gate management more efficient in this way], the vast majority of users starts by getting to know ScadaBR/ScadaLTS at the university level with an arduino uno in hand trying to connect to the supervisory with a usb cable even using the serial modbus protocol.

The defense of the correctness of this protocol is based on facilitating the adoption of this wonderful system by new users and giving more connection options for some equipment, such as multimeters and input PLC's without the need for additional hardware.

fernandoborbabritos, 2021-05-11:
Goodnight everyone. I agree with what was explained by Fabio, Wagner and Ourique. It is very important to have the serial Modbus connection function available, mainly to study and practice. In my case I have several devices (Powermeters, Temperature Controller, Power Factor Controller, and others) with serial connection in my Laboratory of the School (currently I teach Scada), and it is very important to have this feature. In ScadaBR1.1 we already know that Modbus Serial does not work, but we hoped that in the LTS version this would be corrected. I hope this detail can be solved and have the serial connection. Thank you.

Limraj, 2021-05-12:
We covered the topic in the team because there is so much interest in this feature, we decided to tackle it in the coming iterations, the latest it should be in 2.9.0. Greetings to everyone, thank you for your votes.

dev-pts, 2021-07-01:
I also encountered this problem on Debian 10 after installing librxtx-java packet. In my case, it was solved by setting an appropriate LD_LIBRARY_PATH before starting the tomcat:
```
export LD_LIBRARY_PATH=/usr/lib/jni/
./bin/startup.sh
```

Limraj, 2022-06-22:
Nobody needs it anymore?

Limraj, 2022-06-23:
docker tag: pr-2266
In the docker-compose.yml file, I added device sharing for docker:
```
devices:
   - /dev/:/dev/
```
As a result, I have access to ports from the docker.

For all:
1. You must have a docker-compose installed.
2. Download file and uzip: [docker-compose.zip](https://github.com/SCADA-LTS/Scada-LTS/files/8964400/docker-compose.zip)
3. Create dir example scadalts and move file *.yml
4. Open terminal in scadalts dir and execute command:
```
 docker-compose up database
 docker-compose up scadalts
```

Limraj, 2022-06-24:
Hello @fabiodurao,
Did you manage to launch the docker with ports and test something?

Best wishes.

Limraj, 2022-10-07:
@fabiodurao 
You just have this port written down and cannot be used in the test. You can choose a different port and click scan. Best to just create a Data Source, attach a point and test it on the point.

fabiodurao, 2022-10-07:
I tested serial modbus communication on ScadaBR 1.0, an old version before the bug was accidentally added to version 1.1 (which was the branch for Scada-LTS that inherited the bug) and I tested it on ScadaBR 1.2 (version created with some java, tomcat and with this bug fixed). I tested in those versions that I know to see if there was no problem in the communication on the PLC side.

My test on Scada-LTS was not successful because I forgot to do the procedure mentioned above to release the ports, I will redo it and post the result.

closed 2022-10-31 by Limraj

JoseVinicius1998, 2023-03-24:
I would like to give my feedback about the modbus serial in Scada-lts, version 2.7.2. 
I got to test the Modbus Serial via a docker container, using as host an Ubuntu 22.04. I had success only reading registers and coils, but i can't do writing operations. 
I am not sure if I'm doing something wrong, then I prefered to relate this situation here! 

Thanks for the good work guys!

JoseVinicius1998, 2023-03-27:
Sorry. I noticed that a made a mistake.
The silly mistake of not considering subtracting the register address value by one... 
Actually, the Modbus serial is working perfectly and I also can do commands!

Limraj, 2023-03-27:
Hi @JoseVinicius1998,
Okey. Sorry if that sounded rude. If something happens, be sure to let us know.

Regards,
Kamil Jarmusik