SCADA-LTS/Scada-LTS #1701: Scada LTS freezes after 2 hours of normal operation

cristianmichi, 2021-05-13:
**Describe the bug**
Hello everyone, I am using scadaLTS to communicate with reclosers using the DNP3 protocol, I have more than 10 slaves working, through gsm routers and a VPN through the cellular network.

The problem is that my software (scadaLTS) freezes after approximately 2 hours of communicating correctly with all my slaves. I need help on how to solve this.

I am using a SQL database sending to the HeidiSQL software.

**To Reproduce**
Steps to reproduce the behavior:
1. I create in a normal way the DNP3 TCP-IP slaves
2. I assign the corresponding parameters to the protocol, source address, slave address, ip address, times, port, etc. for each of my devices.
2. Then I create each datapoint in the normal way
3. The software communicates with all the devices in a normal way, but as time passes, the time for updating the values of each device increases, until finally they are all completely frozen (approximately 2 hours).
4. The error is solved by restarting the tomcat, that is, I stop the tomcat (pressing the stop button) and then again I press start, and everything returns to normal.

**Expected behavior**
The normal behavior is to get the information all the time from all my DNP3 slaves without freezing the software.

**Screenshots**
In this image we can see all my parameterized DNP3 devices.
[image]

**Desktop (please complete the following information):**
 - OS: [Windows 10 64 bit]
 - Browser [chrome]
 - Version [Scada-LTS - 2.5]

**Additional context**
Investigating the problem I have tried with different SQL database managers such as: xampp, HeidiSQL, ScadaLTS running on Ubuntu Server with a pre-installed SQL database manager, etc. but the behavior is the same.

I also tried to get good results by modifying the values of maxActive, maxWait, maxIdle but I can't solve the problem.
Please help! Thank you.

Limraj, 2021-05-13:
Hey, can you send me the logs from the time of the crash?

cristianmichi, 2021-06-01:
1. This is a screenshot of the logs at the time of the problem. Apparently it does have something to do with memory: "This is very likely to create a memory leak":
[image]

4. jdk/jre: jre-8u72
mysql server: XAMPP 8.0.2 - phpMyAdmin
tomcat: Apache Tomcat 7.0 Tomcat7
system: windows10 Pro - RAM: 16 gb - 64 bits 
chrome:  91.0.4472.77
ScadaLTS v2.5.1

5. https://github.com/SCADA-LTS/Scada-LTS/releases
6. Yes, I have also used ScadaBR 1.0 and I have the same problem, I have also used other database managers such as HeidiSQL.
I'll run it in Firefox and see how it behaves.
What could be done to fix the memory issue?
Thanks!

Limraj, 2021-06-04:
Hey, this reported problem could be related to many things, including a failed app initialization. Moreover, after the date it seems that these are not logs from version 2.5.1, but rather old ones?
My suggestion:
1. would be to clear the mango.log file, then run the application in version 2.5.1; 
2. next define the "internal datasource", and watchlist with datapoins; 
3. run VisualVM to track CPU and memory usage, there will also be information with what parameters jvm was fired with. (https://visualvm.github.io/download.html)
4. when the problem occurs, please send the entire mango.log and flyway.log file (tomcat_home/logs), you can email: kamil.jarmusik@gmail.com, we will establish further steps.
Greetings.

closed 2023-03-26 by Limraj